natario1/CameraView #284: NPE while doing setPreviewTexture

jishindev, 2018-10-12:
- [x] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

Getting this when running the app on an AndroidThings device with API 27.
How is the Camera1 api's being used here? Sorry that I could not file the issue as you had requested, time constraints.
```
018-10-12 18:55:04.415 I/CameraPreview: onSurfaceAvailable: w= 1920 h= 984
2018-10-12 18:55:04.418 I/Camera1: onSurfaceAvailable: Size is 1920x984
2018-10-12 18:55:04.420 I/Camera1: onSurfaceAvailable: Inside handler. About to bind.
2018-10-12 18:55:04.421 I/Camera1: bindToSurface: Started
2018-10-12 18:55:04.427 E/AndroidRuntime: FATAL EXCEPTION: CameraViewController
    Process: com.creoit.android.visitorcounter, PID: 2422
    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.setPreviewTexture(android.graphics.SurfaceTexture)' on a null object reference
        at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:120)
        at com.otaliastudios.cameraview.Camera1.access$500(Camera1.java:27)
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:78)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:63)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2018-10-12 18:55:04.432 E/CameraController: uncaughtException: Unexpected exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.setPreviewTexture(android.graphics.SurfaceTexture)' on a null object reference
    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.setPreviewTexture(android.graphics.SurfaceTexture)' on a null object reference
        at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:120)
        at com.otaliastudios.cameraview.Camera1.access$500(Camera1.java:27)
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:78)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:63)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2018-10-12 18:55:04.433 I/CameraController: destroy: state: STATE_STARTED
2018-10-12 18:55:04.435 I/CameraController: stopImmediately: State was: STATE_STARTED
2018-10-12 18:55:04.436 I/Camera1: onStop: About to clean up.
2018-10-12 18:55:04.438 W/Camera1: onStop: Clean up. Returning.
2018-10-12 18:55:04.439 I/CameraController: stopImmediately: Stopped. State is: STATE_STOPPED

```

natario1, 2018-10-12:
If you could post more logs before these ones. And set CameraLogger to verbose. It's really strange!

natario1, 2018-10-12:
Try also using a different Facing value before starting the camera. If using back, try front.

jishindev, 2018-10-12:
CameraLogger is set to VERBOSE.

```
I/MainActivity: openPreview() called
2018-10-12 18:55:04.038 W/OrientationEventListener: Cannot detect sensors. Not enabled
2018-10-12 18:55:04.040 I/CameraController: Start: posting runnable. State: STATE_STOPPED
2018-10-12 18:55:04.043 I/CameraController: Start: executing. State: STATE_STOPPED
2018-10-12 18:55:04.045 I/CameraController: Start: about to call onStart() STATE_STARTING
2018-10-12 18:55:04.053 I/CameraController: Start: returned from onStart(). Dispatching. STATE_STARTING
2018-10-12 18:55:04.054 I/CameraCallbacks: dispatchOnCameraOpened null
2018-10-12 18:55:04.075 D/OpenGLRenderer: HWUI GL Pipeline
2018-10-12 18:55:04.130 W/CameraView: preview: isHardwareAccelerated: true
2018-10-12 18:55:04.179 W/OrientationEventListener: Cannot detect sensors. Not enabled
2018-10-12 18:55:04.222 W/CameraView: onMeasure: surface is not ready. Calling default behavior.
2018-10-12 18:55:04.262 W/CameraView: onMeasure: surface is not ready. Calling default behavior.
2018-10-12 18:55:04.319 I/OpenGLRenderer: Initialized EGL, version 1.4
2018-10-12 18:55:04.319 D/OpenGLRenderer: Swap behavior 1
2018-10-12 18:55:04.320 W/CameraView: onMeasure: surface is not ready. Calling default behavior.
2018-10-12 18:55:04.332 W/CameraView: onMeasure: surface is not ready. Calling default behavior.
2018-10-12 18:55:04.415 I/CameraPreview: onSurfaceAvailable: w= 1920 h= 984
2018-10-12 18:55:04.418 I/Camera1: onSurfaceAvailable: Size is 1920x984
2018-10-12 18:55:04.420 I/Camera1: onSurfaceAvailable: Inside handler. About to bind.
2018-10-12 18:55:04.421 I/Camera1: bindToSurface: Started
```

natario1, 2018-10-12:
Try with the other Facing value. I have already fixed this (opening camera that does not exist) in v2 (unreleased yet)

romariomkk, 2019-02-21:
@natario1 please explain what changes have been done for fixing this exact issue, as understanding is lost with the introduction of v2