natario1/CameraView #287: does CameraUtils.decodeBitmap take front/back camera into aacount?

suarezu, 2018-10-16:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. using the front camera to tacke picture, and CameraUtils.decodeBitmap to show what captured
2. the captured looks 180-degree horizontally flipped
3.

#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

natario1, 2018-10-16:
No & this is intended behavior for now. But it's easy for you to do this, take a look at StackOverflow.
You just can't use CameraUtils.decodeBitmap(), sorry.

suarezu, 2018-10-17:
as CameraUtils.decodeBitmap() comments, it processes the orientation of the taken image; android bitmapfactory doesnt, how to combine them?
besides, even if picture can be worked around, how about video? why it is intended? is it possible to expose this parameter for user to make the choice?

natario1, 2018-10-17:
For two reasons,
- flipping is hard and expensive in terms of performance
- it's questionable wether we should flip. If you lift your left hand, the output picture will have you lifting your left hand. And this is also the Android default.

To flip pictures you would have to add some matrix.setScale() inside decodeBitmap (look on StackOverflow). So you can just copy paste the decodeBitmap code in your app and add that line

suarezu, 2018-10-17:
yes, I have checked the built-in camera apps on some devices at my hands, some flipped and some didnot, I really recommend leaving the choice of difficulty and expense to the user of the library :D

digirishab, 2019-01-01:
Yes same, Please add this feature.

closed 2019-01-11 by natario1

Rishabh-sx, 2019-05-06:
Hi, I am still facing this issue. When I take picture/video from front camera. The output of picture & video both is getting flipped. 

I have used version - 2.0.0-beta04

PR #360 (2019-01-11): Flip snapshots, add PictureResult.getFacing and VideoResult.getFacing
- Fixes #15 
- Fixes #287 
- Tests: yes