natario1/CameraView #289: Problem in HTC One after capture video video shown with green color not real capture video

gamalsebaee18, 2018-10-16:
Great library ! 
but i have problem in HTC One after capture video video shown with green color not real capture video
![screenshot_2018-10-16-14-21-56](https://user-images.githubusercontent.com/37069135/47016243-5b594a00-d14f-11e8-8d0b-4578ff708e23.png

natario1, 2018-10-22:
You can't right now and I don't see this as coming anytime soon, sorry!

shivkoshti, 2018-11-27:
have you done with this issue ?

natario1, 2018-12-23:
I am not against it but not planning to work on it either. Feel free to do it if you want, but please do so in the v2 branch as it changes everything about video recording.

tolew1, 2019-06-07:
@natario1 it is actually a very good feature that not many have. OpenCamera has that functionality and it's a very good lib. But it can be a bit slow for certain things if you need fast shots and init.

sathishmscict, 2021-01-27:
> @natario1 it is actually a very good feature that not many have. OpenCamera has that functionality and it's a very good lib. But it can be a bit slow for certain things if you need fast shots and init.

Yes, this might be make this library more special and unique feature. Even WhatsApp, Signal and Telegram not support this feature.