natario1/CameraView #292: startAutoFocus(x,y) sometimes does not work

wosiu, 2018-10-17:
- [ x] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

`startAutoFocus()` does not always work. E.g. when it is called from inside cameraListener sometimes only onFocusStart is called but autofocus is somehow killed or not triggered at all, and onFocusEnd also is not called. That would be nice to have it working so that we can achieve following cyclic workflow:

1. startAutofocus
2. onFocusEnd is triggered -> call captureSnapshot
3. onPhotoTaken is triggered -> process image, call startAutofocus
4. goto 2

In code:
```
...
cameraView.addCameraListener(new CameraListener() {
                @Override
                public void onCameraOpened(CameraOptions options) {
                    cameraView.startAutoFocus(cameraView.getWidth() / 2, cameraView.getHeight() / 2);
                }

                @Override
                public void onPictureTaken(byte[] jpeg) {
                    processOrWhatever(jpeg);
                    cameraView.startAutoFocus(cameraView.getWidth() / 2, cameraView.getHeight() / 2);
                }

                @Override
                public void onFocusEnd(boolean successful, PointF point) {
                            cameraView.captureSnapshot();
                 }
);
```

Sometimes it iterates like this few times and stops.

In implementation of startAutoFocus in Camera1.java there is:

                // TODO this is not guaranteed to be called... Fix.
                mCamera.autoFocus(new Camera.AutoFocusCallback() {
                    @Override
                    public void onAutoFocus(boolean success, Camera camera) {
                        // TODO lock auto exposure and white balance for a while
                        mCameraCallbacks.dispatchOnFocusEnd(gesture, success, p);
                        mHandler.get().removeCallbacks(mPostFocusResetRunnable);
                        mHandler.get().postDelayed(mPostFocusResetRunnable, mPostFocusResetDelay);
                    }
                });

Does the "TODO" refers to the problem I've just described?

natario1, 2018-12-16:
It does, sometimes `Camera.AutoFocusCallback` will not call `onAutoFocus`.
This is a Camera1 limitation.

The solution would be to add a timout after which we call `onAutoFocus` ourselves. But this require a lot of complex logic to handle all the scenarios. So it's still a TODO.

natario1, 2019-07-11:
This was fixed.

closed 2019-07-11 by natario1