natario1/CameraView #293: need help

Rupeshrajsingh, 2018-10-19:
if i want to take picture by volume UP or volume Down button,
so what i do..?

natario1, 2018-10-19:
Try asking on StackOverflow!

JeremeLau, 2018-10-23:
Just set a onKeyDown listener and listen 2 KeyEvent.KEYCODE_VOLUME_DOWN or KeyEvent.KEYCODE_VOLUME_UP, then do the TAKEPICTURE action

natario1, 2018-12-16:
I think it would be cool if the library listened to these two and treated them as a `Gesture` so you can assign `GestureAction`s.

MoustafaElsaghier, 2019-09-10:
you can try this 
```
mCameraView.setOnKeyListener(new View.OnKeyListener() {
            @Override
            public boolean onKey(View v, int keyCode, KeyEvent event) {
                if(keyCode ==  KeyEvent.KEYCODE_VOLUME_DOWN || keyCode == KeyEvent.KEYCODE_VOLUME_UP){
                    // capture image code here (        mCameraView.takePictureSnapshot();)      
                }
                return false;
            }
        });
```

SweetD3v, 2023-07-06:
Working on that. Will create a pull request when done.