natario1/CameraView #295: Null Bitmap from `FrameProcessor`

the-dagger, 2018-10-22:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Add the FrameProcessor to the CameraView
2. Use BitmapFactory to create a Bitmap from the received ByteArray
3. Perform some computation on the Bitmap

#### Expected behavior:
The code should work normally

#### Actual behavior:
The app crashes with a NullPointer Exception on the Bitmap

#### Code Snippet:
```
cameraView.addFrameProcessor { frame ->
                val bitmap = BitmapFactory.decodeByteArray(it.data,0,it.data.size)
                bitmap.width         //NullPointer here   
}
```

#### Crash Logs
```
 java.lang.NullPointerException: Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()' on a null object reference
```

natario1, 2018-10-22:
AFAIK decodeByteArray reads jpeg arrays, so it can't read the frame processor data which is ImageFormat.NV21 as explained in the docs.

the-dagger, 2018-10-22:
Ah I see.
Thanks for this info, I might have skipped over it in the docs, thanks!

closed 2018-10-22 by the-dagger

Priyanka-Mohanty, 2019-03-12:
how did you fix that issue...I have the same issue