natario1/CameraView #298: what is the code to save the captured picture to storage

jom16antonio, 2018-10-24:
How do you save picture to a file
how to set the filename

natario1, 2018-10-25:
Try to ask on StackOverflow, the lib will give you a jpeg byte array

Divay-Parashar, 2018-10-25:
hey this Code works for me hope this will work for you
 cameraView.addCameraListener(new CameraListener() {
@Override
            public void onPictureTaken(byte[] jpeg) {
                super.onPictureTaken(jpeg);
                 SavePhotoTask(jpeg);
            }
        });
public void SavePhotoTask(byte [] jpeg)
{
   File imagesFolder = new File(Environment.getExternalStorageDirectory(), "msApp");
            imagesFolder.mkdirs();

            final File photo= new File(imagesFolder, "Yourimagename.jpg");
            try
            {
                FileOutputStream fos=new FileOutputStream(photo.getPath());
                fos.write(jpeg[0]);
                fos.close();
            }
          catch(Exception e)
          {
                 
           }
}

jom16antonio, 2018-10-27:
This worked great. thanks
Hope this gets added to documentation

ifanfaizal, 2018-11-03:
I followed 

> hey this Code works for me hope this will work for you
> cameraView.addCameraListener(new CameraListener() {
> @override
> public void onPictureTaken(byte[] jpeg) {
> super.onPictureTaken(jpeg); [...]

I followed this, but no pictures were saved 😢 

NB: I've set permission in Manifest too

Divay-Parashar, 2018-11-16:
Hi. Ifan, please check your app read and write storage and camera permission in your device setting and make sure these permissions are set on for storing and capturing an image. If this does not work please send me the code I'll take look into this.

closed 2018-12-16 by natario1

gabriel-TheCode, 2020-01-25:
THE METHOD PROVIDED IS :
```java
   @Override
        public void onPictureTaken(@NonNull PictureResult result) {
            super.onPictureTaken(result);
    
        }  
```

NOT

```java
   @Override
        public void onPictureTaken(byte[] jpeg) {
            super.onPictureTaken(result);
        }
        
```

4nku5h, 2020-09-30:
there is no method named     -> public void onPictureTaken(byte[] jpeg)

4nku5h, 2020-09-30:
cameraView.addCameraListener(new CameraListener() {
            @Override
            public void onCameraOpened(@NonNull CameraOptions options) {
                super.onCameraOpened(options);
            }

            @Override
            public void onCameraError(@NonNull CameraException exception) {
                super.onCameraError(exception);
            }

            @Override
            public void onPictureTaken(@NonNull PictureResult result) {
                Toast.makeText(getActivity(), result.getData().toString(), Toast.LENGTH_SHORT).show();
                SavePhotoTask(result.getData());

                super.onPictureTaken(result);
            }
        });

 public void SavePhotoTask(byte [] jpeg)
    {
        File imagesFolder = new File(Environment.getExternalStorageDirectory(), "msApp");
        imagesFolder.mkdirs();

        final File photo= new File(imagesFolder, "Yourimagename.jpg");
        try
        {
            FileOutputStream fos=new FileOutputStream(photo.getPath());
            fos.write(jpeg[0]);
            fos.close();
        }
        catch(Exception e)
        {

        }
    }

Nazir02, 2022-03-11:
btn = findViewById(R.id.takePicture);
        camera = findViewById(R.id.camera);
        camera.setLifecycleOwner(this);
        camera.setFacing(Facing.FRONT);

        btn.setOnClickListener(v -> {
            camera.takePicture();
            camera.getResources();
        });
        camera.addCameraListener(new CameraListener() {
            @Override
            public void onPictureTaken(@NotNull PictureResult result) {
                // A Picture was taken!

                SavePhotoTask(result.getData());
                super.onPictureTaken(result);
            }
            // And much more
        });
    }

    public void SavePhotoTask(byte[] jpeg) {
        File imagesFolder = new File(Environment.getExternalStorageDirectory(), "msAPP");
        imagesFolder.mkdirs();

        final File photo = new File(imagesFolder, "1.jpg");
        try {
            Toast.makeText(this, jpeg.toString(), Toast.LENGTH_SHORT).show();
            FileOutputStream fos=new FileOutputStream(photo.getPath());
            fos.write(jpeg[0]);
            fos.close();
        } catch (Exception e) {
            Toast.makeText(this, "noSave", Toast.LENGTH_SHORT).show();
        }
    }

I followed this, but no pictures were saved cry

NB: I've set permission in Manifest too