natario1/CameraView #300: Memory leak

WrathOfFlame, 2018-10-26:
- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

Detected memory leak while trying to go back and forth between two activities. Returning to the activity that contains CameraView component adds up about 6mb each time. The bug is reproduced on 4 different devices.

#### Detailed steps to reproduce the issue:

1. Start a new project. Add `CameraView` definition to an `.xml` file. Add `onResume()`, `onPause()` and `onDestroy()` methods to the calling activity. Add a button, and set OnClick listener, as follows:
```
            @Override
            public void onClick(View v)
            {
                Intent intent = new Intent(getApplicationContext(), DetailsActivity.class);
                startActivity(intent);
            }
```
2. Create another activity. You can use a simple button to call `finish();` inside of onClickListener
3. Use Android Profiler tool to watch how memory consumption grows each time the CameraView.start() method is called

#### Expected behavior:
no memory leak
#### Actual behavior:
About 6 MB are allocated each time the CameraView.start() method is called

WrathOfFlame, 2018-10-26:
Well, I tried it right now. The leak is there. I can go up from 12 up to 40 MB, when eventually clean up is triggered

natario1, 2018-10-26:
Are you able to see what kind of data is this memory holding? Lots of folks here are using leak detection libraries and we got no reports, so it might be just the gc running when it decides to.

WrathOfFlame, 2018-10-26:
Yeah, it seems so. Is there anything I can do about that?

jom16antonio, 2018-10-27:
Is it possible that this leak could lead to ANR?

closed 2018-12-16 by natario1