natario1/CameraView #303: onPictureTaken is called very late after calling cameraView.capturePicture();

mohitrajput987, 2018-10-29:
onPictureTaken is called very late after calling cameraView.capturePicture();
Till then, camera screen looks steady.

Namazed, 2018-10-31:
Hi, @mohitrajput987 
Which resolution you choose for your picture?
If I right by default biggest resolution.

mohitrajput987, 2018-11-02:
Yes, default resolution.

Namazed, 2018-11-02:
@mohitrajput987 
I recommend you don't use default, because the biggest resolution is long processed (devices can have 4k resolution). I tested and decided that optimal resolution is FullHD(1920x1080).
The Telegram app choose this resolution too.

fahime-ghasemi, 2018-11-16:
@Namazed I have this problem too, How can I change the default resolution? How can I create mPictureSizeSelector?

fahime-ghasemi, 2018-11-19:
@mohitrajput987 Do you find a solution for this problem?

Namazed, 2018-11-19:
Hi, @fahime-ghasemi 
For example: https://gist.github.com/Namazed/3bbecb8d25243a6088e578c515598a17

Kishor-Giri, 2018-12-06:
Hello @Namazed How can i set the output picture size equal to device height and device width?

Namazed, 2018-12-06:
Hi, @Kishor-Giri 
I answered you in your issue.

closed 2018-12-16 by natario1