natario1/CameraView #307: cameraView  Not full screen

greadking, 2018-11-01:
*cameraView  Not full screen .
Prerequisites:*in the fragment, cametaView - width and heigth set at wrap_content

<com.otaliastudios.cameraview.CameraView
        android:id="@+id/cameraView"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content" />

mobile phone brand is huawei(Quidway)

ifanfaizal, 2018-11-03:
have you changed the layout_height and layout_width to match_parent?

greadking, 2018-11-03:
set at match_parent Of course it's good，But there will be problems with window and photo size

greadking, 2018-11-03:
> have you changed the layout_height and layout_width to match_parent?
set at match_parent Of course it's good，But there will be problems with window and photo size

huseynlilkin, 2018-11-29:
I also had same problem, changing layout width and height to match_parent solved it. Also you may want to add android:theme="@style/Theme.AppCompat.NoActionBar" your activity declaration at manifest in order to hide action bar.

closed 2018-12-16 by natario1

sandrocsimas, 2020-05-17:
I setted the layout width and height to match_parent but I'm still seeing a white bar on the right side of the screen. Anyone can help me?