natario1/CameraView #308: How can i use both camera on app?

VishalChauha, 2018-11-01:
i am trying to run both camera at same time like half of the screen is from rear camera and other half is from front camera but unfortunately i got result from only rear camera, front camera shows me black screen.

Do anybody have any idea how can i do that?
I have 10.OR.E phone with android version 8.0, Nexus 5X with Android 8.1 and Oppo phone with lollipop version.

Is there any solution that i can make both camera work on all devices?

Thanks.

agalfioni, 2018-11-05:
With new changes in release version 1.6.0 that can be possible? I also need to record at the same time from frontal and back camera.

Namazed, 2018-11-06:
@agalfioni 
You can see all changes here:
https://github.com/natario1/CameraView/blob/v1.6.0/CHANGELOG.md

natario1, 2018-12-16:
It's not possible, sorry.

closed 2018-12-16 by natario1