natario1/CameraView #309: Photo is No Square!

xuanxws, 2018-11-02:
- [ ] *I am using the latest version   1.6.0

1. I want to take a square photo,But when i set code below

       <com.otaliastudios.cameraview.CameraView
                    android:id="@+id/camera_view"
                    android:layout_width="match_parent"
                    android:layout_height="match_parent"
                    android:keepScreenOn="true"
                    app:cameraPictureSizeAspectRatio="1:1" />

   High profile mobile phone wil take a square photo,but some low profile phone will not take a square photo.

#### Expected behavior:take a square photo

#### Actual behavior:will not take a square photo

#### Screenshots:

#### Interesting logs:

Namazed, 2018-11-02:
Hi, @xuanxws 
I think, you have this problem because cameraView select best resolution for this ratio, if on device doesn't have resolution for this ratio, then cameraView select biggest resolution by default.
I can wrong, but it all depends on the hardware camera. 
You need write your custom ```SizeSelector```.

Namazed, 2018-11-06:
Your selector does not take into account what the hardware camera may not has resolution for this case:
```java
SizeSelectors.and(ratio, dimensions)
```
Because you choose only one size, your device's width, and you use maxWidth and maxHeight twice.
```java
SizeSelectors.maxWidth(ScreenUtils.getScreenWidth(getContext()));
SizeSelectors.maxHeight(ScreenUtils.getScreenWidth(getContext()));
SizeSelectors.maxWidth(ScreenUtils.getScreenWidth(getContext()));
SizeSelectors.maxHeight(ScreenUtils.getScreenWidth(getContext()));
```
I recommend you use step for ratio, because the hardware camera may not has square ratio 
or its square ratio is greater than your screen width in this situation you should use step for your width.

If your device camera doesn't have square resolution you should cut your image.

Maybe owner of this library can to help you more.

natario1, 2018-12-16:
As @Namazed said :-)

closed 2018-12-16 by natario1