natario1/CameraView #323: Incorrect picture is taken in "flat" landscape.

billschauweker, 2018-11-14:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x ] *I am using the latest version*
- [x ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Start the demo app.
2. Choose a page or picture that you like in landscape; prop it up like on a music stand.
3. Orient the device to landscape. Take a picture.
4. Everything is fine - picture fills the screen.
5. Lay the page/picture flat on a desktop.
6. Orient the device to landscape.
7. Tilt the device so that the page/picture again fills the viewfinder (in landscape).
8. Snap a picture.
9. Notice that the picture is a portrait.

#### Expected behavior:
I would expect to be able to take a landscape picture even if the item being viewed is laying horizontal.
#### Actual behavior:
As described above, the picture comes out "portrait".
#### Screenshots:

None yet, but I would be happy to send some.

#### Interesting logs:

I'm just starting to learn my way around the code, and am not sure yet where to focus. My initial guess is that the code may be grabbing the wrong orientation value when a picture is snapped.

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

wosiu, 2018-11-30:
orientation is in the EXIF info. Check this one:
https://github.com/natario1/CameraView/blob/d0bb8cd31cf4fe263be8c95dec314fbdf8a9994c/cameraview/src/main/utils/com/otaliastudios/cameraview/CameraUtils.java#L131

natario1, 2018-12-16:
Orientation is not reliable when holding the device horizontally. I think that #290 fixed this as best as we can.

closed 2018-12-16 by natario1