natario1/CameraView #328: FrameProcessor, MATCH_PARENT and cropOutput

alexsullivan114, 2018-11-19:
Just wanted to start off by saying this is a *fantastic* library. I'm a huge fan, and it's made a huge difference in the amount of time it takes for me to spin up any interactions with the (brutal) camera API.

That being said, I'm running into an issue that I'd love some help troubleshooting.

I'm currently building an app that includes a barcode scanning feature. There's a screen in the app where the top third or so of the screen is a camera view and below that there's a list of items that need to be scanned.

Right now the actual camera view has its height and width both set to `MATCH_PARENT`. My understanding is that leads to a cropped preview image, which makes total sense since it distorts the aspect ratio of the image.

I'm using a `FrameProcessor` to pass each frame over to a firebase barcode scanning library to detect if a barcode is in the picture. That's also working great.

The only issue is that the frame that we're passing into the `FrameProcessor` isn't the same frame that the user is actually seeing, so it's quite difficult for the user to focus their phone in on a single barcode, because while it may appear that the user is only looking at one barcode through the scanner there may actually be another barcode above or below that one that the user can't see because it's clipped by the preview.

I've tried setting the `cropOutput` property on the scanner view with no luck - it sounds like it's not supported for videos and I imagine supporting it for the frame processor would encounter similar speed and memory issues.

Can anyone think of a solution to this particular problem?

wosiu, 2018-11-30:
`cropOutput` works for `createSnapshot` and `createPicture` only AFAIR. however it takes some time. 
After you get frame in FrameProcessor you have to preprocess it (crop based on the ROI you need) bedore you forward it ML kit.
In Android there is `BitmapRegionDecoder` that I use on photo taken by `createSnapshot`. Not sure if it's gonna work on the frame from FrameProcessor, as the latter has different image format AFAIR (I might be wrong).

alexsullivan114, 2018-12-04:
Alright, so I've ended up with the following (unfortunate) code:

```kotlin
val image = YuvImage(frame.data, frame.format, size.width, size.height, null)
val out = ByteArrayOutputStream()
image.compressToJpeg(
  Rect(0, 0, size.width, size.height),
   50,
   out
)
val bitmap = BitmapFactory.decodeByteArray(out.toByteArray(), 0, out.size())
val matrix = Matrix()

matrix.postRotate(frame.rotation.toFloat())
val rotated = Bitmap.createBitmap(bitmap, 0, 0, bitmap.width, bitmap.height, matrix, true)
val cropRect = computeCrop(rotated.width, rotated.height, previewViewAspectRatio)
val crop =  Bitmap.createBitmap(rotated, cropRect.left, cropRect.top, cropRect.width(), cropRect.height())
```

Which is a bummer. Basically the logic is as follows:

1. Convert the provided byte array to a YuvImage so I can...
2. Get a JPG out of it so I can...
3. Get a bitmap out of it so I can...
4. Rotate the bitmap since the image comes rotated. I need to rotate it (as far as I can tell) so I can...
5. Crop the bitmap. This cropped bitmap is the same view as what shows up in my preview, which is great.

The downside, of course, is that all the above operations are _super_ expensive. I feel like I should be able to skip steps 2, 3, 4, and 5 and just provide a cropping rectangle to the original YuvImage compress to jpg call, but unfortunately I wasn't able to get a valid cropping rectangle.

If anyone has any better ideas for how to do this I'm all ears.

GuanacoDevs, 2018-12-04:
I think that at the moment that's the way to go. I have my app deployed using Camera2 for devices not LEGACY and LIMITED and Camera1 with the FUNTASTIC **_CameraView_**. (CameraView is the only thing I needed, but worked too hard to get it working with Camera2 that I didn't wanted to just droped my code)
Something that may help is to choose the size of the frame taken, depending on the requirements of your app. I choose the minimum as 480px(width) and choose from there the minimum area, The I do the same as you, rotate if needed, cut an area for another reading and black out the areas I do not need from the original picture.
I'm reading bingo cards and these are the samples of the final product before processing it with MLKit.
[image]
[image]
All in all takes +/- 1/2 second from taking the frame to reading the card, **CameraView**, is faster, like really fast.
I do blackout the sides, because I noticed that I got better readings using the original width than croping the area of interest(I tried that).
So my advice would be to make sure you get the best smallest picture side and then process it as you have described(maybe without cropping).

[This is a video](https://www.youtube.com/watch?v=WrSsQnHH074) of what I'm reading, that video is with camera2, CameraView is maybe 2X faster.

alexsullivan114, 2018-12-04:
Great information @GuanacoDevs thanks. I'll try lowering the size and see if that makes a big performance difference.

closed 2018-12-04 by alexsullivan114