natario1/CameraView #331: Front camera focus not working

Mritunjay10, 2018-11-23:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1.
2.
3.

#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

Mritunjay10, 2018-11-23:
When I switch to front camera focus is not working but on rear view, the focus is working.
please help me out.

Thanks, for the great library!

natario1, 2018-12-16:
Focus might be not supported by your device in that configuration. There's nothing we can do from our side in this case.

Thanks!

closed 2018-12-16 by natario1