natario1/CameraView #333: Hi Video Stops recording when we switch the camera facing in between .

jindalRahul, 2018-11-24:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Open the CameraView for video recording 
2. While video is recording change the camera facing
3. Video will stop recording . And record only till the point you have changed the camera facing

#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

natario1, 2018-12-16:
This is intended behavior, sorry.

closed 2018-12-16 by natario1

jindalRahul, 2018-12-17:
Thanks for the information. But is there any way that the video from both camera can be recorded in a single file ?

natario1, 2018-12-17:
I guess you can merge them later, but not with this library