natario1/CameraView #334: hello I want to take a full screen picture....how can i set the output image height and width equals to device height and width

Kishor-Giri, 2018-12-05:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1.
2.
3.

#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

Namazed, 2018-12-06:
Hi, @Kishor-Giri 
Use your SizeSelector or resize your ready picture.
You need understand, that if hardware camera doesn't support resolution of your device screen, you can't do it, but you can find nearest resolution for your device resolution with SizeSelector.
The default SizeSelector choose biggest resolution for the output picture.

Namazed, 2018-12-06:
@Kishor-Giri 
I think, you can use Bitmap.createScaledBitmap or something else, maybe third party library (Picasso).

closed 2018-12-16 by natario1