natario1/CameraView #339: Pinch zoom flickers the camera view

Amrutha-shet, 2018-12-20:
I have used Camera View in my layout and also overridden onMesure() to handle the aspect ratio to fit my screen. 

The issue is every time I pinch zoom there is some kind of flicker.

natario1, 2018-12-20:
Does this happen with the demo app?

Amrutha-shet, 2018-12-20:
Nope this does not happen with demo app

natario1, 2018-12-20:
Are you able to spot the difference? Is this due to your onMeasure implementation?

Amrutha-shet, 2018-12-21:
Match parent is not fitting the screen due to difference in aspect ratio.

natario1, 2018-12-21:
Match parent does exactly that, unless your device / activity is not hardware accelerated, in which case the lib unfortunately does not work well.

closed 2019-01-17 by stale[bot]

Pinkal7600, 2019-04-29:
I am facing the same issue with my app with the camera view and also with the demo app. While I perform pinch to zoom, camera preview is stacking and take some time to stable preview. I have tested two different devices. Please check below Device details and attached video.

1. Device details:
Name : Nokia 6.1
OS : Android 9.0

2. Device details:
Name : Motorola One Power
OS : Android 9.0

Video link: https://drive.google.com/file/d/1o61peUf3xreqEhrvbhh0wLBAW2LVQfBd/view?usp=sharing

Pinkal7600, 2019-04-29:
As I checked in other Android devices with OS 6.0 and 7.0 its working fine. So, I think an issue is generated in Android 9.0 only. Please check this issue.