natario1/CameraView #340: Integration with firebase MLKIT

sshrabah, 2018-12-20:
### An integration guide with firebase MLKIT

Hello Everyone, I am working on a project that uses the firebase machine learning framework MLKIT for an application that uses a barcode reader. However, the cameraView proposed by the framework is terrible. It doesn't work on all devices. 

Because I am not that familiar with CameraView. i wonder if someone can write a guide on how to integrate it with MLKIT barcode scanner. 

Tnak you in advanced.

natario1, 2018-12-20:
I'm not a MLKIT expert so I'll leave this one to someone else. You will need to use a FrameProcessor in CameraView and pass data to ML

GuanacoDevs, 2018-12-20:
@sshrabah CameraView makes it incredible easy to work with MLKit due to its FrameProcessor.
My integration guide would be something like:
- Set the minimal requirements for CameraView.
- Set a Frame Processor
- Use the FrameProcessor to pass the frames to MLKit

Just make sure to follow up on [Performance Tips](https://firebase.google.com/docs/ml-kit/android/recognize-text##performance_tips) from MLKit.

I built an OCR App succesfully with Camera2 before finding out about CameraView, with CameraView I get half the lines of codes than with CameraView and I am newbie.

@natario1 It is my understanding that using the FrameProcessor to process anything like MLKit is not being done in the Main Thread, is that correct?

natario1, 2018-12-20:
Correct

bipinvaylu, 2019-01-01:
Hey @natario1 I have developed face detection using ML Kit and Camera preview frame processing using FrameProcesser. But as and when I start video capturing, Frame processor callback stopped. I am not sure why it is stopped, Is that intentionally or It is due to some incorrect setup at my side?

natario1, 2019-01-03:
This is how Camera1 works, sorry. You can't receive callbacks while capturing video. In v2 you will be able to capture a `videoSnapshot` which does not block the frame processor.

sshrabah, 2019-01-04:
I actually succeeded to do what I wanted using FrameProcessor. This thing rocks. thanks @natario1

natario1, 2019-01-04:
Cool! Yes, videoSnapshot should be working.

closed 2019-01-04 by natario1

bipinvaylu, 2019-01-04:
Hey @sshrabah,  It would be nice if you can share your thoughts on how you figure it out. I am wondering if you haven't faced [this](https://github.com/natario1/CameraView/issues/347) issue?

Priyanka-Mohanty, 2019-03-13:
how to implement it,, for smiling face detection..can you please any sample ?

amit2273, 2020-03-16:
@sshrabah 

Is your ML Kit working fine with camera View on All devices.
My Bounding box is getting shifted to the left in some of the devices while others it works fine. Im already using frame processor .

Can you help me with this if you have faced the same issue?
also @natario1   can you suggest something on this