natario1/CameraView #341: Access denied finding property "camera.hal1.packagelist"

Anikeshpatel, 2018-12-29:
W/CameraViewContr: type=1400 audit(0.0:51835): avc: denied { read } for name="u:object_r:camera_prop:s0" dev="tmpfs" ino=5417 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:camera_prop:s0 tclass=file permissive=0
I/zygote64: Compiler allocated 6MB to compile void android.view.ViewRootImpl.performTraversals()
E/libc: Access denied finding property "camera.hal1.packagelist"
W/CameraViewContr: type=1400 audit(0.0:51836): avc: denied { read } for name="u:object_r:camera_prop:s0" dev="tmpfs" ino=5417 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:camera_prop:s0 tclass=file permissive=0

closed 2019-01-25 by stale[bot]

dnhyde, 2019-06-05:
This still happens on some devices (Xiaomi Redmi 5 plus here)

crifan, 2020-08-05:
the error 

`E/libc: Access denied finding property "camera.hal1.packagelist"`

is caused by previous warning:

`W/CameraViewContr: type=1400 audit(0.0:51835): avc: denied { read } for name="u:object_r:camera_prop:s0" dev="tmpfs" ino=5417 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:camera_prop:s0 tclass=file permissive=0
I/zygote64: Compiler allocated 6MB to compile void android.view.ViewRootImpl.performTraversals()`

more detailed explanation can refer [another post's answer](https://stackoverflow.com/questions/59533551/access-denied-finding-property-persist-vendor-log-tel-dbg/63259127#63259127)