natario1/CameraView #343: How to disable camera Shutter sound ?

ManuMobsuccess13, 2019-01-02:
Hello I noticed that "the Camera1 engine will always play shutter sounds regardless of this flag"
But actually I'm on Camera2 because I'm using Android Huawei Tablet with API 26 and I STILL have the shutter sound when I'm taking photos and videos
I added "cameraView.setPlaySounds(false);" to my code at the initialization and "app:cameraAudio="off"" to my XML

Any solutions ? Thanks

natario1, 2019-01-03:
It depends on the device apparently. The Camera1 comment applies to your device, we use the Camera1 APIs regardless of device API level. Nothing we can do.

jiaoya, 2020-06-19:
I have encountered this problem too. Can't I solve it?

jiaoya, 2020-06-19:
I also encountered this problem, can't it be solved?

jiaoya, 2020-06-19:
I used mCameraView.setEngine(Engine.CAMERA2);
  mCameraView.setPlaySounds(false); there is still sound

Bielak2000, 2023-02-24:
I have this problem too for Engine.Camera2.