natario1/CameraView #346: aspect ratio for VideoView

kirogluerhan, 2019-01-03:
Is it possible to give aspect ratio for VideoView? I give "app:cameraPictureSizeAspectRatio="3:4" to CameraView, it changes the screen size while recording. But after that previewing video is not 3:4 ratio. 

Thanks for your work, great repository.

borichellow, 2019-02-21:
The same story for me.
in xml i do:
```
    app:cameraVideoSizeAspectRatio="1:1" 
    app:cameraPictureSizeAspectRatio="1:1"
```
as a result: image is squared, but video is not.
Tried to do the same programmatically 
```
val ratio = SizeSelectors.aspectRatio(AspectRatio.of(1, 1), 0f)
cameraView.setPictureSize(ratio)
cameraView.setVideoSize(ratio)
```
result is the same.

Can somebody help: how to set video-ratio correctly?

dokiKit, 2019-03-12:
I think the it depends on what the ratio the sensor is providing. if the camera sensor does not provide a 1:1 ratio, setting this will not have effect. You can still use takeVideoSnapshot for it though. Correct me if I am wrong.