natario1/CameraView #347: Taking video snapshot not recording proper video

bipinvaylu, 2019-01-04:
### Describe the bug
- When I am recording a video using `takeVideoSnapshot` method. It's getting recorded but somehow it's not storing properly. It's showing very high total duration like `117:22:25` and also not working properly on QuickTime player as well. 

FYI: I have a requirement like processing the frame while recording the video, To achieve this, I had copied camera module to my demo project (somehow v2-beta release is not working via gradle).

- CameraView version: v2-beta (v2 branch)
- Reproducible in official demo app: not tried, but I have created one demo [here](https://github.com/bipinvaylu/FrameTracker/tree/v2-video-snapshot).
- Device / Android version: Nokia 6.1 Plus, API 28

### Expected behavior
App should record video in properly, show proper duration and able to play by any video player.

### Screenshots
I have created one video with the steps in my demo app, Please find [here](http://recordit.co/9DxENCCGZB).

### Logs
```01-04 16:13:26.954 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598606952 processors: 1
01-04 16:13:26.985 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598606985 processors: 1
01-04 16:13:27.018 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607018 processors: 1
01-04 16:13:27.049 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607048 processors: 1
01-04 16:13:27.081 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607080 processors: 1
01-04 16:13:27.119 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607118 processors: 1
01-04 16:13:27.152 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607151 processors: 1
01-04 16:13:27.183 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607182 processors: 1
01-04 16:13:27.220 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607218 processors: 1
01-04 16:13:27.256 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607255 processors: 1
01-04 16:13:27.286 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607285 processors: 1
01-04 16:13:27.318 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607316 processors: 1
01-04 16:13:27.358 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607357 processors: 1
01-04 16:13:27.388 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607387 processors: 1
01-04 16:13:27.420 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607419 processors: 1
01-04 16:13:27.460 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607459 processors: 1
01-04 16:13:27.489 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607488 processors: 1
01-04 16:13:27.521 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607519 processors: 1
01-04 16:13:27.566 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607563 processors: 1
01-04 16:13:27.597 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607596 processors: 1
01-04 16:13:27.626 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607623 processors: 1
01-04 16:13:27.655 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607654 processors: 1
01-04 16:13:27.688 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607687 processors: 1
01-04 16:13:27.721 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607720 processors: 1
01-04 16:13:27.753 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607752 processors: 1
01-04 16:13:27.787 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607786 processors: 1
01-04 16:13:27.826 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607824 processors: 1
01-04 16:13:27.866 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607865 processors: 1
01-04 16:13:27.900 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607900 processors: 1
01-04 16:13:27.935 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607935 processors: 1
01-04 16:13:27.950 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607949 processors: 1
01-04 16:13:27.977 28740-28786/test.example.frametracker D/SnapshotVideoRecorder: CameraLogger - Creating frame encoder. Rotation: 0
01-04 16:13:27.983 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598607982 processors: 1
01-04 16:13:27.995 28740-28786/test.example.frametracker D/WorkerHandler: CameraLogger - get: Creating new handler. EncoderEngine
01-04 16:13:28.016 28740-28874/test.example.frametracker D/MediaEncoder: CameraLogger - Computed a max duration of 9.2233725E15
01-04 16:13:28.030 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608029 processors: 1
01-04 16:13:28.061 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608060 processors: 1
01-04 16:13:28.085 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608084 processors: 1
01-04 16:13:28.120 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608119 processors: 1
01-04 16:13:28.155 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608154 processors: 1
01-04 16:13:28.186 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608185 processors: 1
01-04 16:13:28.206 28740-28874/test.example.frametracker D/EglBaseSurface: CameraLogger - ROTBUG eglSurfaceWidth: 1080 eglSurfaceHeight: 1874
01-04 16:13:28.216 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608216 processors: 1
01-04 16:13:28.257 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608256 processors: 1
01-04 16:13:28.286 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608286 processors: 1
01-04 16:13:28.325 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608324 processors: 1
01-04 16:13:28.353 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608352 processors: 1
01-04 16:13:28.410 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608410 processors: 1
01-04 16:13:28.438 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608437 processors: 1
01-04 16:13:28.472 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608470 processors: 1
01-04 16:13:28.488 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608487 processors: 1
01-04 16:13:28.544 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608540 processors: 1
01-04 16:13:28.582 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608580 processors: 1
01-04 16:13:28.592 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598608592 processors: 1
[... 144 lines cut ...]
01-04 16:13:33.379 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598613378 processors: 1
01-04 16:13:33.413 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchFrame: 1546598613411 processors: 1
[... cut ...]
01-04 16:13:34.273 28740-28765/test.example.frametracker D/Camera1: CameraLogger - onStop: Clean up. Returning.
01-04 16:13:34.274 28740-28765/test.example.frametracker D/CameraController: CameraLogger - Stop: returned from onStop(). Dispatching.
01-04 16:13:34.274 28740-28765/test.example.frametracker D/CameraCallbacks: CameraLogger - dispatchOnCameraClosed
01-04 16:13:35.568 28740-28740/test.example.frametracker D/Camera1: CameraLogger - onSurfaceDestroyed
```

### APK
Github-Repo: https://github.com/bipinvaylu/FrameTracker/tree/v2-video-snapshot

bipinvaylu, 2019-01-05:
@natario1 Thanks for fixing the issue #348. I have tried latest demo app and found this video snapshot issue. Here, Please find the recorded [video file](https://www.dropbox.com/s/pz4jgmu5ulmq0r4/video.mp4?dl=0).

Please let me know if you need any other details from my side.

natario1, 2019-01-05:
The fact that the DropBox player keeps spinning without playing, means that the snapshot is broken or is it just my internet connection?

Thank you, great report!

bipinvaylu, 2019-01-05:
Nope, Your internet connection is good enough to play the video,  Seems Snapshot is broken atm. Please play the video from here: http://recordit.co/9DxENCCGZB.

natario1, 2019-01-05:
Yes I have seen that one! But it would be useful to have the actual snapshot MP4 that the lib recorded, so it can be inspected. Though I am nowhere near a video encoding expert

bipinvaylu, 2019-01-05:
Yeah, That is the same video which I have shared via Dropbox link. I hope you can able to download, if not please let me know. FYI, The same issue is happening in the current demo app as well.

bipinvayalu, 2019-01-08:
Hey, @natario1 Any update on this issue? Let me know if you want any input from my end or if you can guide me then I can try to fix it in the library itself.

bipinvaylu, 2019-01-09:
Ah, Actually I am stuck at this point for implementation. Please find recorded video file metadata: https://www.get-metadata.com/result/7ed6c838-c0c9-42f0-97b7-feaf829f3547. Here, I am adding the question which I have asked on StackOverflow. If you can provide some help/insight about the same. https://stackoverflow.com/questions/54091648/camera-2-api-to-preview-record-and-extract-the-preview-frame-to-firebase-face-d. Thanks for your hardwork about this library.

bipinvaylu, 2019-01-11:
##Video Metadata
[image]

##Audio Metadata
[image]

I hope it will helpful for you to investigate. Let me know if you need any other details.

bipinvaylu, 2019-01-11:
@natario1 Please find the detailed log file here: 
[CameraView-Logs.txt](https://github.com/natario1/CameraView/files/2749553/CameraView-Logs.txt)

natario1, 2019-01-28:
@bipinvaylu could you try the fix/video-snapshot-frames branch?

It does not address this issue specifically, but it changes a few things about snapshot recording.

natario1, 2019-01-31:
@eVerup can you try that branch and let me know?

The poor quality is kind of the point of snapshots, meant for small social-quality media. You can improve the video resolution by using setPreviewSize.

eVerup, 2019-01-31:
Thanks for your fast answer! I'm actually working to have a basic quality on the video. For a 5 seconds filming (same environment), the size is about 20MB (2048x1152). With Snapshot video, it is 300Kb, but very poor, bad duration showing, and with errors.

How can i have something better, with a duration max of 10 seconds in my app ? Did i have to change Bitrate, or something else ?

Thanks again and sorry for my bad english, i'm a french entrepreneur. :)

natario1, 2019-01-31:
@eVerup If quality matters, don't use snapshots. For your case I would suggest calling `cameraView.setVideoSize()` to choose a smaller size (2048x1152 is a lot) and recording the video with `cameraView.takeVideo()`.

RAN3000, 2019-02-23:
I had a similar problem too. Video snapshot quality was very poor (you could see big squares). When recording with audio the video snapshot had some errors (black regions). 
Switching to the `fix/video-snapshot-frames` branch solved the quality issues and most of the video errors, some errors are still there at the start of the video.

natario1, 2019-02-23:
Thanks @RAN3000 , I was waiting to find time to write tests about it before merging, but apparently I don't have any. Will merge that now.

What are the errors at the start of the video?
Did you manage to have pictures working in #353 ?

RAN3000, 2019-02-23:
No worries.
I played a little more with it. The errors at the start of the video I was talking about are visual glitch on the first frame. Though I noticed that they only appear on the first time the video is reproduced, so it not a big deal. Thanks for the work!

Yes, I managed to get the picture working yesterday. There's some stuff to be fixed, I hope that by tomorrow I'll have something decently working.

natario1, 2019-03-01:
I'll consider this fixed for now. Thanks.

closed 2019-03-01 by natario1

BlindJerobine, 2019-04-02:
I have exactly the same problem with my Pixel 2XL. It's working fine on Emulator and on other devices. It's about exactly the same as bipinvaylu posted in his video.