natario1/CameraView #348: Demo app: video recording not working (crash)  - error message: "Got CameraException #5"

andreififiita, 2019-01-04:
### Describe the bug
Video capture in demo app is not working. I get "Got CameraException #5" toast message and crash.

- CameraView version: 2
- Reproducible in official demo app: yes
- Device / Android version: Galaxy J7 Android 8.1.0

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open demo app
2. Click edit settings icon and switch to video mode
3. Click record video icon

### Expected behavior
Should start recording video

### Logs
2019-01-04 14:51:40.744 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@68b1ca[CameraActivity]: ViewPostIme pointer 0
2019-01-04 14:51:40.814 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@68b1ca[CameraActivity]: ViewPostIme pointer 1
2019-01-04 14:51:40.888 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setup
2019-01-04 14:51:40.890 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setVideoSource(1)
2019-01-04 14:51:40.897 18794-18794/com.otaliastudios.cameraview.demo D/InputTransport: Input channel constructed: fd=122
2019-01-04 14:51:40.899 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@66aeebb[Toast]: setView = android.widget.LinearLayout{fe6cfd8 V.E...... ......I. 0,0-0,0} TM=true MM=false
2019-01-04 14:51:40.899 18794-18794/com.otaliastudios.cameraview.demo V/Toast: Text: Seco in android.widget.Toast$TN@1aa7f31
2019-01-04 14:51:40.901 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setAudioSource(0)
2019-01-04 14:51:40.902 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setVideoFrameRate(30)
2019-01-04 14:51:40.902 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setVideoSize(1920, 1080)
2019-01-04 14:51:40.903 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setVideoEncoder(2)
2019-01-04 14:51:40.903 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setParameter()
2019-01-04 14:51:40.903 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setParameter()
2019-01-04 14:51:40.903 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@66aeebb[Toast]: dispatchAttachedToWindow
2019-01-04 14:51:40.904 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setParameter()
2019-01-04 14:51:40.904 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setAudioEncoder(3)
2019-01-04 14:51:40.905 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: setParameter()
    
    --------- beginning of crash
2019-01-04 14:51:40.909 18794-18813/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewController
    Process: com.otaliastudios.cameraview.demo, PID: 18794
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.io.File.getAbsolutePath()' on a null object reference
        at com.otaliastudios.cameraview.FullVideoRecorder.start(FullVideoRecorder.java:82)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:666)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-01-04 14:51:40.910 18794-18813/com.otaliastudios.cameraview.demo E/CameraController: uncaughtException: Unexpected exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.io.File.getAbsolutePath()' on a null object reference
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.io.File.getAbsolutePath()' on a null object reference
        at com.otaliastudios.cameraview.FullVideoRecorder.start(FullVideoRecorder.java:82)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:666)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2019-01-04 14:51:40.910 18794-18813/com.otaliastudios.cameraview.demo I/CameraController: destroy: state: STATE_STARTED
2019-01-04 14:51:40.911 18794-18813/com.otaliastudios.cameraview.demo I/CameraController: stopImmediately: State was: STATE_STARTED
2019-01-04 14:51:40.912 18794-18813/com.otaliastudios.cameraview.demo I/Camera1: onStop: About to clean up.
2019-01-04 14:51:40.912 18794-18813/com.otaliastudios.cameraview.demo I/MediaRecorderJNI: stop
2019-01-04 14:51:40.913 18794-18813/com.otaliastudios.cameraview.demo E/MediaRecorder: stop called in an invalid state: 4
2019-01-04 14:51:40.915 18794-18813/com.otaliastudios.cameraview.demo W/FullVideoRecorder: stop: Error while closing media recorder. Swallowing java.lang.IllegalStateException
    java.lang.IllegalStateException
        at android.media.MediaRecorder._stop(Native Method)
        at android.media.MediaRecorder.stop(MediaRecorder.java:1400)
        at com.otaliastudios.cameraview.FullVideoRecorder.stop(FullVideoRecorder.java:116)
        at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:264)
        at com.otaliastudios.cameraview.CameraController.stopImmediately(CameraController.java:217)
        at com.otaliastudios.cameraview.CameraController.destroy(CameraController.java:155)
        at com.otaliastudios.cameraview.CameraController.uncaughtException(CameraController.java:116)
        at java.lang.Thread.dispatchUncaughtException(Thread.java:1955)
[... 16 lines cut ...]
2019-01-04 14:51:40.961 18794-18794/com.otaliastudios.cameraview.demo D/InputEventReceiver: channel 'b3e10d2 Toast (client)' ~NativeInputEventReceiver.
2019-01-04 14:51:40.975 18794-18794/com.otaliastudios.cameraview.demo D/InputTransport: Input channel destroyed: fd=122
2019-01-04 14:51:40.992 18794-18794/com.otaliastudios.cameraview.demo D/InputTransport: Input channel constructed: fd=87
2019-01-04 14:51:40.994 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@34a7a33[Toast]: setView = android.widget.LinearLayout{e63c1f0 V.E...... ......I. 0,0-0,0} TM=true MM=false
2019-01-04 14:51:40.995 18794-18794/com.otaliastudios.cameraview.demo V/Toast: Text: Hot  in android.widget.Toast$TN@530e369
2019-01-04 14:51:40.996 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@34a7a33[Toast]: dispatchAttachedToWindow
2019-01-04 14:51:41.016 18794-18794/com.otaliastudios.cameraview.demo V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
2019-01-04 14:51:41.018 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@34a7a33[Toast]: Relayout returned: old=[0,0][0,0] new=[269,1636][810,1752] result=0x7 surface={valid=true 3470632960} changed=true
2019-01-04 14:51:41.022 18794-18794/com.otaliastudios.cameraview.demo D/ViewRootImpl@34a7a33[Toast]: MSG_RESIZED_REPORT: frame=Rect(269, 1636 - 810, 1752) ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1
2019-01-04 14:51:41.088 18794-18813/com.otaliastudios.cameraview.demo D/Camera: app passed NULL surface
2019-01-04 14:51:41.090 18794-18813/com.otaliastudios.cameraview.demo I/Camera1: destroyCamera: Clean up. Releasing camera.
2019-01-04 14:51:41.369 18794-18813/com.otaliastudios.cameraview.demo I/Camera1: destroyCamera: Clean up. Released camera.
2019-01-04 14:51:41.373 18794-18813/com.otaliastudios.cameraview.demo W/Camera1: onStop: Clean up. Returning.
2019-01-04 14:51:41.373 18794-18813/com.otaliastudios.cameraview.demo I/CameraController: stopImmediately: Stopped. State is: STATE_STOPPED
2019-01-04 14:51:41.377 18794-18794/com.otaliastudios.cameraview.demo D/AndroidRuntime: Shutting down VM
2019-01-04 14:51:41.378 18794-18794/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.otaliastudios.cameraview.demo, PID: 18794
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.io.File.getAbsolutePath()' on a null object reference
        at com.otaliastudios.cameraview.FullVideoRecorder.start(FullVideoRecorder.java:82)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:666)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)

bipinvaylu, 2019-01-04:
Yeah, I am facing the same issue with today's last demo app.

natario1, 2019-01-04:
Should be fixed now! Please reopen in case it isn't

closed 2019-01-04 by natario1