natario1/CameraView #349: App crashs when starting taking video

ManuEpi, 2019-01-04:
Hello , first of all , great lib thanks

### Describe the bug

When I click on my button I start capturing video , but sometimes , like once every 10 , it stop recording instantly and throw "E/MediaRecorder: stop failed: -1007"

Is there really nothing we can do about this ? Will it be fixed or handle in v2 ?
This bug apparently happens when : you Start and stop the recorder too fast . 

Take a look at this blog it can be useful : http://davehiren.blogspot.com/2017/01/android-media-recorder-stop-failed-1007.html

I read that it happens sometimes with MediaRecorder but I really need to handle or fix this crash
Actually this annoys me , because even if it does not happen often , it mess my project a bit.

- CameraView version: *1.6.1*
- Reproducible in official demo app: *yes/no*
- Device / Android version: *ASUS P00C, API 23*
- Device / Android version: *Samsung Galaxy Tab 2, API 16*

### Expected behavior
It should start a video and recording it for about few seconds instead of crash

ManuEpi, 2019-01-09:
I "solved" this problem by adding a countdown , so what happen right now : 
My Video recorder will start after a 3 seconds countdown.

For example : 
I start taking picture , and if I want to take a video right after , I will click on Video and then a 3 second countdown will start and after that my video will start.
So that let time to video record to take place
( Sorry for my English )

closed 2019-02-06 by stale[bot]

HBiSoft, 2019-04-23:
@natario1 I also commented in another question earlier today, as you know from that comment the exception is still present in v2 - https://github.com/natario1/CameraView/issues/425#issuecomment-485647640