natario1/CameraView #350: Camera preview is black after upgrading to 2.0.0-beta01

tyczj, 2019-01-04:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: *2.0.0-beta01*
- Device / Android version: *Pixel 3, API 28*

### Logs
W/Activity: Can request only one set of permissions at a time
V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 12
V/FA: Activity paused, time: 1127243865
V/FA: Activity resumed, time: 1127243888

Just updated from 1.6.1 to the new 2.0.0-beta01 and now everytime my activity starts I get this log message and the camera just stays black

W/Activity: Can request only one set of permissions at a time
V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 12
V/FA: Activity paused, time: 1127243865
V/FA: Activity resumed, time: 1127243888

this is my activity

```
class BarcodeScanningActivity : AppCompatActivity(),FrameProcessor {

    private lateinit var detector:FirebaseVisionBarcodeDetector

    override fun onCreate(savedInstanceState:Bundle?){
        super.onCreate(savedInstanceState)
        setContentView(R.layout.camera_view)

        val options:FirebaseVisionBarcodeDetectorOptions = FirebaseVisionBarcodeDetectorOptions.Builder()
            .setBarcodeFormats(FirebaseVisionBarcode.FORMAT_DATA_MATRIX)
            .build()

        detector = FirebaseVision.getInstance().getVisionBarcodeDetector(options)
        camera.setLifecycleOwner(this)
        camera.addFrameProcessor(this)
        camera.mode = Mode.PICTURE
        camera.facing = Facing.BACK
        camera.flash = Flash.OFF
    }

    override fun onResume() {
        super.onResume()
        camera.open()
    }

    override fun onPause() {
        super.onPause()
        camera.close()
    }

    override fun onDestroy() {
        super.onDestroy()
        camera.destroy()
    }

    override fun process(frame: Frame) {
        val metadata = FirebaseVisionImageMetadata.Builder()
            .setWidth(480)
            .setHeight(360)
            .setFormat(FirebaseVisionImageMetadata.IMAGE_FORMAT_NV21)
            .setRotation(FirebaseVisionImageMetadata.ROTATION_0) // always assuming portrait
            .build()

        val image = FirebaseVisionImage.fromByteArray(frame.data, metadata)

        val result = detector.detectInImage(image)
            .addOnSuccessListener { barcodes ->
                if(!barcodes.isEmpty()){
                    val barcode = barcodes.first()
                    val intent = Intent()
                    intent.putExtra("barcode",barcode.rawValue)
                    setResult(Activity.RESULT_OK)
                    finish()
                }
            }
            .addOnFailureListener {

            }
    }

}
```

natario1, 2019-01-04:
If you are using setLifecycleOwner(this), you must remove `onResume`, `onPause` and `onDestroy` calls. Not needed anymore

tyczj, 2019-01-04:
oh nice!

tyczj, 2019-01-04:
Just removed the onResume/Pause/Destroy and i get a different messages in the logcat

```
2019-01-04 15:45:52.734 1147-2852/? I/ActivityManager: START u0 {act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.packageinstaller cmp=com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity (has extras)} from uid 10276
2019-01-04 15:45:52.770 1147-2876/? I/ActivityManager: START u0 {act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.packageinstaller cmp=com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity (has extras)} from uid 10276
2019-01-04 15:45:52.770 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH ON
2019-01-04 15:45:52.777 30043-30070/jtycz.simple.mesh.connector V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 12
2019-01-04 15:45:52.778 30043-30070/jtycz.simple.mesh.connector V/FA: Activity paused, time: 1141717904
2019-01-04 15:45:52.778 30091-30091/? W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@e4e9850
2019-01-04 15:45:52.803 30043-30070/jtycz.simple.mesh.connector V/FA: Activity resumed, time: 1141717930
2019-01-04 15:45:52.807 1147-1393/? I/ActivityManager: START u0 {act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.packageinstaller cmp=com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity (has extras)} from uid 10276
2019-01-04 15:45:52.808 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH ON
2019-01-04 15:45:52.815 30043-30070/jtycz.simple.mesh.connector V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 12
2019-01-04 15:45:52.815 30043-30070/jtycz.simple.mesh.connector V/FA: Activity paused, time: 1141717942
2019-01-04 15:45:52.816 30091-30091/? W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@d975a26
2019-01-04 15:45:52.835 30043-30070/jtycz.simple.mesh.connector V/FA: Activity resumed, time: 1141717962
2019-01-04 15:45:52.839 1147-2876/? I/ActivityManager: START u0 {act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.packageinstaller cmp=com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity (has extras)} from uid 10276
2019-01-04 15:45:52.840 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH ON
2019-01-04 15:45:52.847 30043-30070/jtycz.simple.mesh.connector V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 12
2019-01-04 15:45:52.847 30043-30070/jtycz.simple.mesh.connector V/FA: Activity paused, time: 1141717974
2019-01-04 15:45:52.848 30091-30091/? W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@1ba9ac
2019-01-04 15:45:52.867 30043-30070/jtycz.simple.mesh.connector V/FA: Activity resumed, time: 1141717994
2019-01-04 15:45:52.870 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH OFF
2019-01-04 15:45:52.875 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH ON
2019-01-04 15:45:52.882 30043-30070/jtycz.simple.mesh.connector V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 16
2019-01-04 15:45:52.883 30043-30070/jtycz.simple.mesh.connector V/FA: Activity paused, time: 1141718010
2019-01-04 15:45:52.884 30091-30091/? W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@2e131f3
2019-01-04 15:45:52.906 30043-30070/jtycz.simple.mesh.connector V/FA: Activity resumed, time: 1141718032
2019-01-04 15:45:52.874 1147-2876/? I/ActivityManager: START u0 {act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.packageinstaller cmp=com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity (has extras)} from uid 10276
2019-01-04 15:45:52.909 1147-2852/? I/ActivityManager: START u0 {act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.packageinstaller cmp=com.google.android.packageinstaller/com.android.packageinstaller.permission.ui.GrantPermissionsActivity (has extras)} from uid 10276
2019-01-04 15:45:52.910 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH ON
2019-01-04 15:45:52.917 30043-30070/jtycz.simple.mesh.connector V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 12
2019-01-04 15:45:52.918 30043-30070/jtycz.simple.mesh.connector V/FA: Activity paused, time: 1141718044
2019-01-04 15:45:52.918 30091-30091/? W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@b0573c8
2019-01-04 15:45:52.919 794-846/? I/CHRE: @ 1141720.625: [AR_CHRE] still: 100
```

Again they just keep repeating.

I even removed `camera.setLifecycleOwner(this)` and put back in the Pause/Resume/Destroy and still did it

tyczj, 2019-01-04:
You can pull my repo if you want

https://github.com/tyczj/SimpleMeshConnector

natario1, 2019-01-05:
You should enable logs using CameraLogger.setLogLevel(VERBOSE) and then post them here so we can see what's wrong.

tyczj, 2019-01-05:
Enabled logging, here is the log starting from when the app first starts

```
2019-01-05 10:05:53.343 16446-16446/jtycz.simple.mesh.connector I/System.out: waiting for debugger to settle...
2019-01-05 10:05:53.373 1147-2557/? V/BackupManagerConstants: getFullBackupRequiredNetworkType(...) returns 2
2019-01-05 10:05:53.373 1147-2557/? V/BackupManagerConstants: getFullBackupRequireCharging(...) returns true
2019-01-05 10:05:53.517 800-2313/? I/WifiHAL: event received NL80211_CMD_VENDOR, vendor_id = 0x1374, subcmd = 0xd
2019-01-05 10:05:53.544 16446-16446/jtycz.simple.mesh.connector I/System.out: waiting for debugger to settle...
2019-01-05 10:05:53.744 16446-16446/jtycz.simple.mesh.connector I/System.out: waiting for debugger to settle...
2019-01-05 10:05:53.944 16446-16446/jtycz.simple.mesh.connector I/System.out: debugger has settled (1442)
2019-01-05 10:05:54.121 623-646/? W/SurfaceFlinger: Attempting to set client state on removed layer: Dim Layer for - mAboveAppWindowsContainers#0
2019-01-05 10:05:54.121 623-646/? W/SurfaceFlinger: Attempting to destroy on removed layer: Dim Layer for - mAboveAppWindowsContainers#0
2019-01-05 10:05:54.124 623-646/? W/SurfaceFlinger: Attempting to destroy on removed layer: aebbcf0 Waiting For Debugger: jtycz.simple.mesh.connector#0
2019-01-05 10:05:54.227 16446-16446/jtycz.simple.mesh.connector W/.mesh.connecto: JIT profile information will not be recorded: profile file does not exits.
2019-01-05 10:05:54.235 16446-16446/jtycz.simple.mesh.connector I/chatty: uid=10278(jtycz.simple.mesh.connector) identical 10 lines
2019-01-05 10:05:54.236 16446-16446/jtycz.simple.mesh.connector W/.mesh.connecto: JIT profile information will not be recorded: profile file does not exits.
2019-01-05 10:05:54.317 16446-16446/jtycz.simple.mesh.connector V/FA: Registered activity lifecycle callback
2019-01-05 10:05:54.351 16446-16446/jtycz.simple.mesh.connector D/FirebaseApp: com.google.firebase.auth.FirebaseAuth is not linked. Skipping initialization.
2019-01-05 10:05:54.356 16446-16446/jtycz.simple.mesh.connector D/FirebaseApp: com.google.firebase.crash.FirebaseCrash is not linked. Skipping initialization.
2019-01-05 10:05:54.356 16446-16446/jtycz.simple.mesh.connector I/FirebaseInitProvider: FirebaseApp initialization successful
2019-01-05 10:05:54.359 16446-16446/jtycz.simple.mesh.connector I/InstantRun: starting instant run server: is main process
2019-01-05 10:05:54.392 16446-16491/jtycz.simple.mesh.connector E/libc: Access denied finding property "vendor.debug.egl.profiler"
2019-01-05 10:05:54.389 16446-16446/jtycz.simple.mesh.connector W/jtycz.simple.mesh.connector: type=1400 audit(0.0:79722): avc: denied { read } for comm=45474C20496E6974 name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=19103 scontext=u:r:untrusted_app:s0:c22,c257,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
2019-01-05 10:05:54.422 16446-16486/jtycz.simple.mesh.connector V/FA: Collection enabled
2019-01-05 10:05:54.422 16446-16486/jtycz.simple.mesh.connector V/FA: App package, google app id: jtycz.simple.mesh.connector, 1:102439305037:android:c9609c419e86c168
2019-01-05 10:05:54.423 16446-16486/jtycz.simple.mesh.connector I/FA: App measurement is starting up, version: 14710
2019-01-05 10:05:54.423 16446-16486/jtycz.simple.mesh.connector I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
2019-01-05 10:05:54.423 16446-16486/jtycz.simple.mesh.connector I/FA: To enable faster debug mode event logging run:
      adb shell setprop debug.firebase.analytics.app jtycz.simple.mesh.connector
2019-01-05 10:05:54.423 16446-16486/jtycz.simple.mesh.connector D/FA: Debug-level message logging enabled
2019-01-05 10:05:54.456 16446-16446/jtycz.simple.mesh.connector V/FA: onActivityCreated
2019-01-05 10:05:54.553 16446-16486/jtycz.simple.mesh.connector V/FA: Connecting to remote service
2019-01-05 10:05:54.587 16446-16446/jtycz.simple.mesh.connector W/.mesh.connecto: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (light greylist, reflection)
2019-01-05 10:05:54.588 16446-16446/jtycz.simple.mesh.connector W/.mesh.connecto: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (light greylist, reflection)
2019-01-05 10:05:54.591 16446-16486/jtycz.simple.mesh.connector V/FA: Connection attempt already in progress
2019-01-05 10:05:54.624 1147-1393/? I/ActivityManager: START u0 {cmp=jtycz.simple.mesh.connector/.ui.barcode.BarcodeScanningActivity} from uid 10278
2019-01-05 10:05:54.625 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH ON
2019-01-05 10:05:54.691 16446-16446/jtycz.simple.mesh.connector D/OpenGLRenderer: Skia GL Pipeline
2019-01-05 10:05:54.693 16446-16486/jtycz.simple.mesh.connector I/FA: Tag Manager is not found and thus will not be used
2019-01-05 10:05:54.705 16446-16486/jtycz.simple.mesh.connector D/FA: Logging event (FE): screen_view(_vs), Bundle[{firebase_event_origin(_o)=auto, firebase_screen_class(_sc)=MainActivity, firebase_screen_id(_si)=-7072495536571621473}]
2019-01-05 10:05:54.760 16446-16486/jtycz.simple.mesh.connector V/FA: Connection attempt already in progress
2019-01-05 10:05:54.760 16446-16486/jtycz.simple.mesh.connector V/FA: Connection attempt already in progress
2019-01-05 10:05:54.762 16446-16486/jtycz.simple.mesh.connector V/FA: Activity resumed, time: 1207719607
[... cut ...]
2019-01-05 10:05:57.049 16446-16486/jtycz.simple.mesh.connector V/FA: Activity resumed, time: 1207721974
2019-01-05 10:05:57.055 793-793/? D/android.hardware.power@1.3-service.crosshatch-libperfmgr: LAUNCH ON
2019-01-05 10:05:57.066 16446-16446/jtycz.simple.mesh.connector I/CameraController: Stop: posting runnable. State: STATE_STOPPED
2019-01-05 10:05:57.066 16446-16486/jtycz.simple.mesh.connector V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 17
2019-01-05 10:05:57.066 16446-16495/jtycz.simple.mesh.connector I/CameraController: Stop: executing. State: STATE_STOPPED
```

khachatryanhead, 2019-01-08:
The same issue on my Nexus 5X (Android 8.1.0).

hardysim, 2019-01-09:
I have the same problem (at least the preview is black after the upgrade to `2.0.0-beta01`.

It seems that `onResume()` if the fragment containing the preview (where I call `setLifecycleOwner(viewLifecycleOwner)`) is called **multiple times** and the only way to stop it is to stop the fragment (I'm using a `DialogFragment` holding the preview).

As soon as I remove `setLifecycleOwner(viewLifecycleOwner)` it works normally (`onResume()` is **only called once**) but the preview is still black.

Using the manual way with `camera.open()` etc. does not help either (no multiple calls but black preview).

natario1, 2019-01-09:
Thanks. To solve this you must add the RECORD_AUDIO permission in the manifest, or use Audio.OFF. I will fix the behavior in the library also.

natario1, 2019-01-09:
The `checkPermissionsManifestOrThrow` method must not check for `mode == Mode.VIDEO`.

PR #356 (2019-01-11): V2 bug fixes
Fixes #350 
Fixes #355 
Fixes #357
closed 2019-01-11 by natario1