natario1/CameraView #352: Error while starting MediaRecorder. Swallowing. java.lang.IllegalStateException

gaurav-emedsim, 2019-01-08:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: V1
- Reproducible in official demo app: no
- Device / Android version: *Samsung S7, API 28*

### To Reproduce
Detail steps to reproduce:
Video recording is throwing IllegalStateException when a video using Exoplayer is playing in another fragment. Once the exception is thrown then from next time it doesn't record any video and start throwing same exception.

### Expected behavior
Should record a video

### Logs
01-08 06:58:34.120 982-17955/? E/QCamera2HWI: stopRecording 7457: sending new param CAM_INTF_PARM_RECORDING_HINT_START OFF
01-08 06:58:34.120 982-5141/? E/MediaPlayerService: WFD client is not created 
01-08 06:58:34.130 982-5141/? E/MPEG4Writer: Stop() called but track is not started
01-08 06:58:34.140 17071-17198/com.smartmirror.emedsim E/MediaRecorder: start failed: -38
01-08 06:58:34.140 17071-17198/com.smartmirror.emedsim E/Camera1: Error while starting MediaRecorder. Swallowing. java.lang.IllegalStateException
    java.lang.IllegalStateException
        at android.media.MediaRecorder._start(Native Method)
        at android.media.MediaRecorder.start(MediaRecorder.java:946)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:645)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:63)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:158)
        at android.os.HandlerThread.run(HandlerThread.java:61)
01-08 06:58:34.140 17071-17198/com.smartmirror.emedsim E/MediaRecorder: stop called in an invalid state: 0
01-08 06:58:34.150 982-18064/? E/OMXNodeInstance: setConfig(de:google.vorbis.decoder, ConfigPriority(0x6f800002)) ERROR: Undefined(0x80001001)

natario1, 2019-01-09:
I don't think you can do this if both CameraView and ExoPlayer are using the audio service. Try using Audio.OFF to record a video without audio, but I'm not sure it will work either.

gaurav-emedsim, 2019-01-09:
Hi,
I fixed the issue. As you have mentioned it was throwing exception because
both were using mic.

Thanks for the reply.

natario1, 2019-01-09:
Cool!

closed 2019-01-09 by natario1