natario1/CameraView #353: Overlay drawing on preview and on picture/video snapshot

RAN3000, 2019-01-08:
### Problem to be addressed
Right now a way for showing an overlay on top of the camera preview is to use some transparent `View` in something like a `FrameLayout`. This has the disadvantage that the overlay cannot be recorded, when snapping a picture it will not appear. 
As for usefulness of this feature, a bunch of scenario that come to my mind are:
- apps that rely on Computer Vision and Machine Learning to show the results of the their algorithms.
- Snapchat/Instagram like video filters
- adding a watermark

### Describe the solution you'd like
This is achievable with the OpenGL based preview. An additional texture could sample the overlay from a `Bitmap` and the fragment shader would take care of layering the camera preview with the overlay on top. By using `takeVideoSnapshot()` or `takePictureSnapshot()` the overlay can be recorded.
I created a very raw demo [here](https://github.com/RAN3000/CameraView) by forking this project: after you run the demo a black preview will appear, when the "edit" button is pressed on the bottom menu the preview will be shown with a squashed red circle on top, the circle is rendered by OpenGL.

This little demo highlights some challenges:
- **Coordination between user's input and camera lifecycle.** (This is the reason for the black screen when starting the preview) Drawing happens at the `EglViewport` level, the overlay texture size must be the same of the input camera texture size. When drawing we need to be sure that the camera, surfaces, OpenGL and all of that are ready. I need to study `CameraView` better to understand if there are some callbacks we can use for this.
- **Performance** (Not really highlighted in the demo, but important) The `RendererThread` should never be blocked by the overlay drawing.
- **Coordination between `EglViewport`s** `GLCameraPreview`, `SnapshotVideoRecorder`and `SnapshotPictureRecorder` use all different instance of `EglViewport`. It would be nice to have for example two `Bitmap`s where one is shared between the three `EglViewport`s and on is specific for the `SnapshotVideoRecorded` so that processing results can be shown on the camera preview and the watermark only appears in the video, along with the processing results.
- **Drawing with the right scales and coordinates** This is tricky, we want the circle to be a circle not an oval. This way of drawing the overlay forces us to use the camera input coordinates and scales, not the output camera preview surface's ones. Different scenario also make one coordinate system more suitable then another: for adding a watermark that would be output video coordinates (== output preview coordinates since using "snapshot"), for a `FrameProcessor`'s results it would be input camera coordinates since they are the ones that the `FrameProcessor` will use to read the frame. In the latter case the user will likely want to transform from one coordinate system to the other to maintain proportions.

My proposal is to create a `OverlayManager` class that holds a queue of drawing operations, the `Bitmap`s and subscribes some modified `EglViewport`s. When it is done drawing on the `Bitmap`s it sends a signal to the `EglViewport`s that it will not touch the `Bitmap`s anymore until every one of them is done updating the textures by sampling the `Bitmap`s.
I'm unexperienced in multi-threading programming so I'm not sure this is the optimal solution.

In terms of API interface what would be the best approach for something like this to be integrated in this library?

natario1, 2019-01-09:
Thank you, this is super cool! Ignoring technical details for now, I would like to understand first what do you have in mind in terms of end API for the developer?

I think you are talking about `Bitmap` because that's what we draw in OpenGL? But a `View` based system would be perfect in my opinion. Do you think this is possible?

```xml
<CameraView
    android:layout_width="match_parent"
    android:layout_height="match_parent">

    <ImageView
        android:layout_isOverlay="true"
        android:layout_showInPreview="false"
        android:layout_showInPicture="false"
        android:layout_showInVideo="true"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="bottom|end"
        android:src="@drawable/watermark" />

    <TextView
        android:layout_isOverlay="true"
        android:layout_showInPreview="true"
        android:layout_showInPicture="true"
        android:layout_showInVideo="false"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="top|center_horizontal"
        android:text="This is a title" />

</CameraView>
```

This would be the XML usage. In Java code, we would have `cameraView.addOverlay(View view, OverlayLayoutParams params)`. The params object extends `FrameLayout.LayoutParams` with that extra boolean info you can see in XML.

The `CameraView` could then host an `OverlayLayout` as the topmost children. When new children are added it should check if it is an overlay or not (has layout_isOverlay=true). If it is, add the child to the `OverlayLayout` instead of itself.

The `OverlayLayout` should have special draw methods (the standard draw, but also drawOnPicture(canvas), and drawOnVideoFrame(canvas)). We can then use `overlayLayout.getDrawingCache()` to get a `Bitmap`. This layout would do its job normally for the preview, then it should be passed to the picture recorder and video recorder.

This would leverage all the Android optimizations about View drawing. Is this possible in your opinion? It is just the coolest API that came to my mind. The key part of course is `view.getDrawingCache()`.

RAN3000, 2019-01-09:
That's a great idea! I think that it can be done, it will be even faster then using a `Bitmap`. 

I found some resources [here](https://github.com/ArtemBogush/AndroidViewToGLRendering) and [here](www.felixjones.co.uk/neo website/Android_View/).
I'll start experimenting.

As for internal API, when working on some already existing class would you prefer me editing it directly or adding a new class?

natario1, 2019-01-09:
Thank you, this would be a great addition!

Feel free to ask here for any doubt. And yes you can edit whatever you want.

I don't know if this overlay layout will need a special onMeasure implementation, and the scale/crop/rotation part will be tricky for sure... But we'll get there! Thanks!

natario1, 2019-01-24:
Thanks @RAN3000 ! I will take a look as soon as I can :-)

No idea about this color issue. It works with videos though? Is it a shader issue?

natario1, 2019-01-24:
This is the compositing shader of a very good library: 

https://github.com/MasayukiSuda/ExoPlayerFilter/blob/master/epf/src/main/java/com/daasuu/epf/filter/GlOverlayFilter.java

Not sure if it helps, I have no GL knowledge, but I see he uses a different approach than yours. You would have to dig into sub/superclasses of that one though.

RAN3000, 2019-01-24:
Yes it works in videos, it probably is the shader but I'm not sure yet. 
I'll take a look at that `mix` function they use in the shader, thanks!

RAN3000, 2019-02-28:
Hey man I bumped in a serious problem. Right now both video and picture recording work perfectly for static `View`s. Animated views (buttons or checkbox when touched, progressBar, ....) are a big problem, it can be summarized as follows: there is no way for a parent to know when the children are invalidated. The parent would be the `OverlayLayout`. 
What happens in practice is that animation run in big steps and visual glitches start to appear. 
I've been digging for a while in the problem and so far I have not been able to find a good solution.

A strategy (which I haven't tested yet) could be to have the user extend the `View`s he needs. Calling some function for us in the `onDraw()` method (called after invalidation) so that we can invalidate the `OverlayLayout`. 

It is ugly but I can't find other solutions.. We could also prepare a ready to use `OverlayImageView` for exampler.

natario1, 2019-03-01:
Hey @RAN3000 don't worry. We will address this in a second moment. 

Would be already cool to get a working solution for still views which addresses all the issues (drawing at the correct place, even when preview is cropping, with the correct scale, correct colors, in preview and both snapshots). No need to complicate this more for now.

Thank you for your work!

RAN3000, 2019-03-02:
Should we let the user handle the overlay orientation with [UI Orientation](https://natario1.github.io/CameraView/docs/more-features.html#ui-orientation)?
Or should we rotate the `OverlayLayout` like the preview?

natario1, 2019-03-02:
I would say let's do nothing & draw, for example, bottom stuff at the View bottom (which is not necessarily the output bottom). Just to keep this as simple as possible for now

RAN3000, 2019-03-02:
Ok, then I'd say it's ready!
I added a static watermark to the example.
[https://github.com/RAN3000/CameraView/tree/overlay-layout](https://github.com/RAN3000/CameraView/tree/overlay-layout)

natario1, 2019-03-02:
@RAN3000 thank you, it's very cool!

I have tried it now and found two issues 

- if I rotate the device 90° and take a snap the watermark is distorted in the picture snap. No idea why
- if I change the preview dimensions while operating, the watermark is often badly positioned or disappears. Also no idea why

I hope you can fix!

RAN3000, 2019-03-08:
Hey!
I should have fixed the problems you mentioned.

Also I added a `invalidateOverlay()` method to `CameraView`, so that also animated views can be supported.
In the [interactive-overlay branch](https://github.com/RAN3000/CameraView/tree/interactive-overlay) I added two `ProgressBar`s to the example, the one in the bottom left corner is drawn on the screen, the one above the watermark on the camera as overlay.
You will notice that the overlay's spinner skips some frames compared to the real one, because of the preview's framerate. It shouldn't be a problem though, what do you think?

natario1, 2019-03-11:
It works very well now!

- I think `invalidateOverlay()` is not needed as anyone can call CameraView.invalidate() with the same effect
- Also don't want to introduce APIs that we might want to remove. Ideally we want this invalidation to be automatic, we'll take care about it later, I'd like to merge this base set of features first
- I have created a `feature/overlays` branch - can you create a PR there so I can comment on code better?

I'll be annoying as there's a lot of stuff that you have done, but I think this is a super cool addition.

natario1, 2019-03-11:
But before going on with the PR I have a big question - why drawing on the preview when we can let Android do this? Views will look better, antialiased... I see no benefit in doing it this way.

The GlCameraPreview code could be much lighter, without callbacks & without creating the extra GL texture. Then the picture recorder and the video recorder could create this new GL texture at any snap(), and release after. They would create a `SurfaceTexture` out of it, then a `Surface` that should somehow be passed to the overlay layout.

- The OverlayLayout , rather than storing the `Surface`, could just have a `draw(Surface)` API and this API could be called by the snapshot recorders, for example (through an interface if you believe so).
- I'm not 100% sure about this, but this could let us call `draw(Surface)` at any snapshot frame, thus providing a smooth animated overlay in snapshot videos.

In my opinion it would also be cleaner if you leave `EglViewport` as it was before, and instead create a new `EglOverlayViewport` class. This way:

- The GlCameraPreview only uses an `EglViewport`
- The recorders use a `EglViewport` (with the same textureId) and also an `EglOverlayViewport` (which they use to create a new textureId -> SurfaceTexture -> Surface -> overlay.draw(Surface)).

I think that if you use 

```
eglViewport.draw()
eglOverlayViewport.draw()
```

The final result should be a composite image similar to the `mix()` command? Am I wrong?

RAN3000, 2019-03-12:
So many new ideas! 
I will have to experiment a little, then I'll get back to you. Everything sounds doable.

natario1, 2019-03-12:
Actually maybe you don't need a `EglOverlayViewport` but just two `EglViewport`s? Since the shader logic is pretty much the same.

Anyway let me know. 👍

RAN3000, 2019-03-22:
Hey! I managed to do it with a single `EglViewport` instance.
Now the code is a little cleaner. I haven't tried yet, but now it should be possible to have an API similar to your [your previous message](https://github.com/natario1/CameraView/issues/353#issuecomment-452830492).
Should I try to implement that before creating the PR?

natario1, 2019-03-22:
Cool! You mean the android:layout_showIn... attributes? I think we can do that later! You can PR when you wish 👍 I would use the feature/overlays branch just so we can merge & complete this without blocking releases on the master branch.

gbcapps, 2019-03-25:
Fix the issue with frame.size.getHeight/getWidth throwing NPEs. Your front preview is also too dark.

natario1, 2019-07-17:
This was merged in #502

closed 2019-07-17 by natario1

lucazin, 2019-09-30:
Hello!

Its possible to add multiple overlays at runtime?  without fix a xml tag ? 
Example:

I choose a bitmap ou drawable from gallery and select the position and its done.. 
or its possible only one overlay ?

RAN3000, 2019-09-30:
Hi, I haven't tried this, but it should work. You can use a `FrameLayout` as overlay, load your bitmaps or drawables into `ImageView`s and put those `ImageView`s in the `FrameLayout`. That way you can achieve "multiple" overlays even if in the end it's always just one overlay.

lucazin, 2019-09-30:
Hi!

So i put a frame layout in the place watermark .. but i have to set match parent and match parent?
if i put 3 images from gallery or from url.. whatever   i will have just one watermark with these 3 images?

Do you know some way to load theses images from gallery and put in framelayout?  and move and change location of screen ? 

Better..
What about edittext inside framelayout ..we can archive this too ?

lucazin, 2019-10-01:
Can u help me?

Im using Camera Engine 1
My device is A70 Samsung.

When u use frontal camera and start recording a video... the activity close

LogCat:

2019-09-30 21:02:03.740 749-20558/? E/AudioFlinger: AudioFlinger_APM :
(AudioFlinger::openRecord) persist.app.permission.monitor is true
2019-09-30 21:02:03.848 939-10359/? E/OMX-VENC: ERROR: unsupported index
1870659589
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error:
ACDB_CMD_GET_AUDPROC_INSTANCE_GAIN_DEP_STEP_TABLE_SIZE Returned = -19
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error: ACDB AudProc vol
returned = -19
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error:
ACDB_CMD_GET_AFE_INSTANCE_COMMON_TABLE_SIZE Returned = -19
2019-09-30 21:02:03.953 696-21445/? E/ACDB-LOADER: Error: ACDB AFE returned
= -19
2019-09-30 21:02:03.968 19535-21430/br.com.oneplace A/libc: Fatal signal 11
(SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f in tid 21430
(VideoEncoder), pid 19535 (br.com.oneplace)
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: *** *** *** *** *** *** ***
*** *** *** *** *** *** *** *** ***
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: Build fingerprint:
'samsung/a70qxtc/a70q:9/PPR1.180610.011/A705MNXXU3ASG7:user/release-keys'
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: Revision: '12'
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: ABI: 'arm64'
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: pid: 19535, tid: 21430,
name: VideoEncoder  >>> br.com.oneplace <<<
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG: signal 11 (SIGSEGV), code 1
(SEGV_MAPERR), fault addr 0x6300750064006f
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x0  0000000000000000  x1
 0000000000000000  x2  0000000000000000  x3  00000007fee8f710
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x4  000000796daad830  x5
 0000000000000005  x6  0000000000000227  x7  7d847b8478857985
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x8  ffffffffffe2d754  x9
 0000000000000001  x10 0000000000000001  x11 7887788877887788
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x12 7d867f857e857d85
 x13 7986778779867b86  x14 0000000000000001  x15 00000007fee8f740
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x16 0000000000000600
 x17 0000007a134de7e8  x18 000000796e5f0788  x19 00000079728da870
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x20 006300750064006f
 x21 00000007fedc5000  x22 0000007975a8b780  x23 0000000000000000
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x24 0000000000000001
 x25 0000000000630069  x26 000000798a027db0  x27 0000000000000001
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     x28 0000000000000001
 x29 000000796e5f09d0
2019-09-30 21:02:04.148 21448-21448/? A/DEBUG:     sp  000000796e5f0590  lr
 000000797fad1e90  pc  000000797fad1498
2019-09-30 21:02:04.163 2168-21409/? E/WakeLock: GCM_HB_ALARM release
without a matched acquire!
2019-09-30 21:02:04.346 21448-21448/? A/DEBUG: backtrace:
2019-09-30 21:02:04.346 21448-21448/? A/DEBUG:     #00 pc 000000000026a498
 /vendor/lib64/egl/libGLESv2_adreno.so
(EsxBltLib::CopyHostPtrBltSw(EsxBltCopyHostPtr const*)+416)
2019-09-30 21:02:04.347 21448-21448/? A/DEBUG:     #01 pc 0000000000261ff0
 /vendor/lib64/egl/libGLESv2_adreno.so
(EsxResource::UpdateSubResourceDirect(EsxContext*, unsigned int, int,
EsxBox const*, EsxSurfaceAccessDesc const*, int)+1112)
2019-09-30 21:02:04.347 21448-21448/? A/DEBUG:     #02 pc 0000000000262d48
 /vendor/lib64/egl/libGLESv2_adreno.so
(EsxTextureObject::PerformShadowCopy(EsxContext*)+608)
2019-09-30 21:02:04.347 21448-21448/? A/DEBUG:     #03 pc 000000000024fa4c
 /vendor/lib64/egl/libGLESv2_adreno.so
(EsxContext::ValidateProgramTexUnitBoundObjs(EsxProgram*, int)+724)
2019-09-30 21:02:04.347 21448-21448/? A/DEBUG:     #04 pc 000000000024c648
 /vendor/lib64/egl/libGLESv2_adreno.so
[... 477 lines cut ...]
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at android.app.LoadedApk.createAppFactory(LoadedApk.java:241)
        at
android.app.LoadedApk.createOrUpdateClassLoaderLocked(LoadedApk.java:770)
        at android.app.LoadedApk.getClassLoader(LoadedApk.java:849)
        at android.app.LoadedApk.getResources(LoadedApk.java:1090)
        at android.app.ContextImpl.createAppContext(ContextImpl.java:2563)
        at
android.app.ActivityThread.handleBindApplication(ActivityThread.java:6144)
        at android.app.ActivityThread.access$1200(ActivityThread.java:240)
        at
android.app.ActivityThread$H.handleMessage(ActivityThread.java:1797)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7094)
        at java.lang.reflect.Method.invoke(Native Method)
        at
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)
2019-09-30 21:02:06.951 21494-21494/? E/[[WEATHER](161085@5)]:
{[289822919A985EADBE56F8FDFF616C43225A3B2279C9A39418D1BC000569344A1DC1508B7A3B3C1BACE5BB711A677EB7E8D339FAD5086C6397E81A4F064CE4E9]}
2019-09-30 21:02:06.954 21494-21494/? E/[[WEATHER](161085@5)]:
{[43EB1C024052B99D53E991930172379630519257355A8D28DD16151FD22314B0705629E7512927CE37D04AEECA3D7536061CCEAADE99499591CE63087FB099F7E59E883323AB4EA0530BD1E3C63000A785DC95222A4B209560BE487EB3F2E4C8CD6194FD67686619D64F1674B0BC25C1052F811A8D7602DA61C4ECD245DEBF24704667FBA64D7703B2354A68D5596929F0D18B4328D2DBC3DEF85BE6FBBB08B9B0E2FBAF31AA352905370D3805293658CE97A5DB66F7A07EB4C7AF467367D7F581054C3E353ED961EB3AF5C14BA6028A]}
2019-09-30 21:02:07.234 620-667/? E/Netd: getNetworkForDns: getNetId from
enterpriseCtrl is netid 0

RAN3000, 2019-10-01:
My suggestion is to first get the "mulitple image loading in `FrameLayout`" working without CameraView. Then, once you get it working as you want, use it as CameraView overlay.
There are plenty resources on the internet for both how to achieve image loading on Android and for `FrameLayout`. 

About the `EditText`: this will be more tricky, therefore I suggest getting the `FrameLayout` with images down first. Later you can take a look [here](https://ran3000.github.io/2019/09/14/cameraview-interactivity.html), for how to add interactivity (e.g. tapping and writing on the `EditText`) to your CameraView overlay.

I don't know about the frontal camera, this issue is not the place to ask about that, sorry.

lucazin, 2019-10-01:
Thanks RAN3000 i will try like you said.

About frontal camera, i open issues and happens with demo and my application.  =/

lucazin, 2019-10-01:
Hi RAN3000 i tested with your strategy about framelayout , but not work like i expected.

i have this 

`        app:cameraPreview="glSurface"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent">

        
        <FrameLayout
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:id="@+id/frame"
            app:layout_drawOnPictureSnapshot="true"
            app:layout_drawOnPreview="true"
            app:layout_drawOnVideoSnapshot="true">

            <com.munon.turboimageview.TurboImageView
                android:id="@+id/turboImageView"
                android:layout_width="match_parent"
                android:layout_gravity="center"
                android:layout_height="match_parent"
                />
            
        </FrameLayout>`

this guy is just a view where u can put objects inside that and move rotate.. in example project works like a charm, its just a view..

but when i put images him .. i cant move on or rotate or something like that..  i think because Cameraview block.

https://github.com/Mun0n/TurboImageProject

RAN3000, 2019-10-01:
[https://ran3000.github.io/2019/09/14/cameraview-interactivity.html](https://ran3000.github.io/2019/09/14/cameraview-interactivity.html) check this out, I wrote an article for when I dealt with something similar: having an interactive view as overlay (so that you can rotate and move it. It should be easily adaptable to your situation.

lucazin, 2019-10-01:
Hi!!

i did what i want with images!
i use some lib and follow your exemple in this site.

My problem now is the crash when recording a snapshot ...with frontal
camera.. :(    i need help

lucazin, 2019-10-01:
RAN3000

I tested with your app (He have cameraview) and happens the same problem..  he close.

I open the app.. change to frontal camera and start recording.. 

the error happen if you repeat this process for 3 or 4 times..     in the cameraview demo happen too..

shivadjango, 2020-05-26:
Hi Ran3000! . I have a concern how can we use this to overlay a watermark to already captured videos i.e Existing video files in memory !