natario1/CameraView #357: java.lang.RuntimeException: setParameters failed on switching to Video mode

mittt, 2019-01-10:
### Describe the bug
java.lang.RuntimeException: setParameters failed on switching to Video mode

- CameraView version: 2.0.0-beta01
- Reproducible in official demo app: Yes
- Device / Android version: Xiaomi Mi A1, Android 8.1

### To Reproduce
1. Build and run the Demo app
2. Switch Mode to Video
3. App Crashes 

### Expected behavior
Not to crash

### Logs
https://gist.github.com/mittt/cb5f4ef87af57e3c0091ebe608420ada

### APK
https://github.com/natario1/CameraView.git/trunk

natario1, 2019-01-10:
Can you fix the link to the APK? Or did you mean to link the CameraView official demo?

mittt, 2019-01-10:
Yes, I was refereeing to the official demo app.

natario1, 2019-01-10:
@mittt I don't have your specific device so I can't help in any way at the moment, the error is not informative. See if this errors disappears when changing some parameters, and maybe we can spot what's wrong if any

mittt, 2019-01-10:
Few more details:

- It was working on the previous version: com.otaliastudios:cameraview:1.6.0 
- I in our app, I tried setting Width and Height  - 1280x720 and the result was the same
- My phone received an update to Android 9.0, today. The demo and my app crashes the same way after the update.

mittt, 2019-01-10:
оptions.getSupportedVideoSizes():
0 = {Size@7583} "960x1280"
1 = {Size@7584} "480x864"
2 = {Size@7585} "480x800"
3 = {Size@7586} "288x352"
4 = {Size@7587} "360x480"
5 = {Size@7588} "720x1280"
6 = {Size@7589} "120x160"
7 = {Size@7590} "480x640"
8 = {Size@7591} "640x480"
9 = {Size@7592} "320x480"
10 = {Size@7593} "144x176"
11 = {Size@7594} "176x144"
12 = {Size@7595} "360x640"
13 = {Size@7596} "2160x4096"
14 = {Size@7597} "2160x3840"
15 = {Size@7598} "480x720"
16 = {Size@7599} "240x320"
17 = {Size@7600} "320x240"
18 = {Size@7601} "1080x1920"

оptions.getSupportedPictureSizes():
0 = {Size@7636} "960x1920"
1 = {Size@7637} "2976x2976"
2 = {Size@7638} "1536x2048"
3 = {Size@7639} "960x1280"
4 = {Size@7640} "768x1280"
5 = {Size@7641} "640x1280"
6 = {Size@7642} "1200x1200"
7 = {Size@7643} "768x1024"
8 = {Size@7644} "480x864"
9 = {Size@7645} "288x352"
10 = {Size@7646} "480x800"
11 = {Size@7647} "360x480"
12 = {Size@7648} "2448x3264"
13 = {Size@7649} "720x1280"
14 = {Size@7650} "1458x2592"
15 = {Size@7651} "3000x4000"
16 = {Size@7652} "1080x1440"
17 = {Size@7653} "2400x3200"
18 = {Size@7654} "480x640"
19 = {Size@7655} "640x480"
20 = {Size@7656} "320x480"
21 = {Size@7657} "1512x2688"
22 = {Size@7658} "360x640"
23 = {Size@7659} "2250x4000"
24 = {Size@7660} "2000x4000"
25 = {Size@7661} "2160x3840"
26 = {Size@7662} "1296x2592"
27 = {Size@7663} "1200x1600"
28 = {Size@7664} "480x720"
29 = {Size@7665} "240x320"
30 = {Size@7666} "320x240"
31 = {Size@7667} "1944x2592"
32 = {Size@7668} "1080x1920"
33 = {Size@7669} "600x800"

natario1, 2019-01-10:
Thanks! Will fix this

closed 2019-01-11 by natario1

mittt, 2019-01-14:
Great Work! Bata02 works on my phone.

commit d20e8350 (2019-01-10): Fix #357
PR #356 (2019-01-11): V2 bug fixes
Fixes #350 
Fixes #355 
Fixes #357
commit a8a4e099 (2019-01-11): V2 bug fixes (#356)

* Fix permissions error

* Fix #355

* Fix #357

* Improve CameraOptions tests