natario1/CameraView #364: Does video recording work for you on Huawei P8?

LauKto, 2019-01-14:
The library works perfect for me on every device, except Huawei P8, where the image is stuck on the last frame of the preview right after I press the button which should start de video recording. The only thing I find interetsting in the logs is this line:

E/Camera: Camera new cameraInitNormal:0

natario1, 2019-01-14:
Doesn't seem so interesting. If you use the demo app you will find more logs.

Also take a look at the `onCameraError` callback.

cezar9, 2019-01-14:
toooooz 🖕🖕💪🏼💪🏼

مُرسل من بريد Yahoo لجهاز iPhone

بتاريخ الاثنين,

cezar9, 2019-01-14:
🖕🖕💪🏼💪🏼

مُرسل من بريد Yahoo لجهاز iPhone

بتاريخ الاثنين,

cezar9, 2019-01-14:
🤞🖕🖕💪🏼💪🏼 ТффZ fe k

مُرسل من بريد Yahoo لجهاز iPhone

بتاريخ الاثنين,

natario1, 2019-01-14:
Are these logs? :-) please stop.

LauKto, 2019-01-15:
Sooo...with our app, only after around 30 seconds after pressing the Start video recording button I get the following logs:

E/Camera: notify error: 1, and: 0
E/Camera: Error 1
E/Camera1: Error inside the onError callback. 1
E/CameraController: uncaughtException: Interrupting thread with state: STATE_STARTED due to CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 1
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 1
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:313)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1241)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:150)
        at android.os.HandlerThread.run(HandlerThread.java:61)
     Caused by: java.lang.RuntimeException: Error inside the onError callback.

I didn't see them until now because I never waited for 30 seconds. I will install today also the demo app and see if I get anything else there

natario1, 2019-01-15:
You mean that you take video, preview stops, then after 30 seconds you get this?

I'm more interested in logs at the moment of the stop.

LauKto, 2019-01-18:
I installed the Demo app, the behavior is intermitent: sometimes it works, sometimes it doesn't (black screen where the preview should be) these are the logs I get when the black screen appears:

01-18 10:30:29.270 10752-10768/com.otaliastudios.cameraview.demo E/Camera: Camera new cameraInitNormal:0
01-18 10:30:29.270 10752-10768/com.otaliastudios.cameraview.demo I/Camera1: createCamera: Applying default parameters.
01-18 10:30:29.283 10752-10768/com.otaliastudios.cameraview.demo I/Camera1: onStart: Ended
01-18 10:30:29.283 10752-10768/com.otaliastudios.cameraview.demo I/CameraController: Start: returned from onStart(). Dispatching. STATE_STARTING
01-18 10:30:29.283 10752-10768/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchOnCameraOpened com.otaliastudios.cameraview.CameraOptions@d8ed9a1
01-18 10:30:29.286 10752-10752/com.otaliastudios.cameraview.demo I/CameraCallbacks: onDeviceOrientationChanged 0
01-18 10:30:29.290 10752-10752/com.otaliastudios.cameraview.demo W/CameraView: preview: isHardwareAccelerated: true
01-18 10:30:29.305 10752-10752/com.otaliastudios.cameraview.demo W/CameraView: onMeasure: surface is not ready. Calling default behavior.
01-18 10:30:29.325 10752-10770/com.otaliastudios.cameraview.demo E/HAL: load: id=gralloc != hmi->id=gralloc
01-18 10:30:29.325 10752-10770/com.otaliastudios.cameraview.demo I/OpenGLRenderer: Initialized EGL, version 1.4
01-18 10:30:29.333 10752-10770/com.otaliastudios.cameraview.demo D/OpenGLRenderer: loaded so path=/system/lib64/libhwuibp.so handle=0x7f8fec65d8
01-18 10:30:29.337 10752-10770/com.otaliastudios.cameraview.demo D/mali_winsys: new_window_surface returns 0x3000
01-18 10:30:29.339 10752-10752/com.otaliastudios.cameraview.demo W/CameraView: onMeasure: surface is not ready. Calling default behavior.
01-18 10:30:29.344 10752-10752/com.otaliastudios.cameraview.demo D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.15, mControlPoint1y = 0.7, mControlPoint2x = 0.2, mControlPoint2y = 0.98
01-18 10:30:29.345 10752-10752/com.otaliastudios.cameraview.demo D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.6, mControlPoint1y = 0.9, mControlPoint2x = 0.8, mControlPoint2y = 1.0
01-18 10:30:29.345 10752-10752/com.otaliastudios.cameraview.demo D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.15, mControlPoint1y = 0.7, mControlPoint2x = 0.2, mControlPoint2y = 0.98
01-18 10:30:29.345 10752-10752/com.otaliastudios.cameraview.demo D/CubicBezierInterpolator: CubicBezierInterpolator  mControlPoint1x = 0.6, mControlPoint1y = 0.9, mControlPoint2x = 0.8, mControlPoint2y = 1.0
01-18 10:30:29.372 10752-10783/com.otaliastudios.cameraview.demo D/mali_winsys: new_window_surface returns 0x3000
01-18 10:30:29.376 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.379 10752-10783/com.otaliastudios.cameraview.demo I/CameraPreview: dispatchOnOutputSurfaceAvailable: w= 1080 h= 1308
01-18 10:30:29.379 10752-10783/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceAvailable: Size is 1080x1308
01-18 10:30:29.381 10752-10768/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceAvailable: Inside handler. About to bind.
01-18 10:30:29.381 10752-10768/com.otaliastudios.cameraview.demo I/Camera1: bindToSurface: Started
01-18 10:30:29.383 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.383 10752-10768/com.otaliastudios.cameraview.demo I/CameraController: computeCaptureSize: result: 3120x4160 flip: true mode: PICTURE
01-18 10:30:29.384 10752-10768/com.otaliastudios.cameraview.demo I/Camera1: size: sizesFromList: [1280x720, 1280x960, 1920x1080, 208x144, 176x144, 352x288, 320x240, 480x320, 640x480, 800x480, 960x544, 960x720, 720x720]
01-18 10:30:29.384 10752-10768/com.otaliastudios.cameraview.demo I/CameraController: size: computePreviewSize: targetRatio: 3:4 targetMinSize: 1080x1308
01-18 10:30:29.387 10752-10768/com.otaliastudios.cameraview.demo I/CameraController: computePreviewSize: result: 1920x1080 flip: true
01-18 10:30:29.387 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.388 10752-10768/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceAvailable Dispatching onCameraPreviewSizeChanged.
01-18 10:30:29.388 10752-10768/com.otaliastudios.cameraview.demo I/CameraCallbacks: onCameraPreviewSizeChanged
01-18 10:30:29.389 10752-10768/com.otaliastudios.cameraview.demo I/CameraPreview: setInputStreamSize: desiredW= 1080 desiredH= 1920
01-18 10:30:29.392 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.396 10752-10768/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceAvailable Starting preview with startPreview().
01-18 10:30:29.397 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.401 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.405 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.410 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
01-18 10:30:29.410 10752-10752/com.otaliastudios.cameraview.demo I/HwSecImmHelper: mSecurityInputMethodService is null
[... cut ...]
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:78) 
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62) 
        at android.os.Handler.handleCallback(Handler.java:743) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:150) 
        at android.os.HandlerThread.run(HandlerThread.java:61) 
01-18 10:30:31.692 10752-10770/com.otaliastudios.cameraview.demo D/mali_winsys: new_window_surface returns 0x3000

natario1, 2019-02-01:
This error gets propagated to `onCameraError` and you should handle it as explained in the docs:

https://natario1.github.io/CameraView/docs/error-handling.html

About why it happens, I don't know.

closed 2019-02-28 by stale[bot]

HBiSoft, 2019-05-20:
@LauKto Did you resolve this issue?

cswangduo, 2019-06-21:
I encounter the same question