natario1/CameraView #367: java.lang.NullPointerException: Attempt to invoke virtual method 'int com.otaliastudios.cameraview.Size.getWidth()' on a null object reference

ZaferAlemdaroglu, 2019-01-16:
### Describe the bug
I use cameraView for Firebase ML Kit (Face recognition and text recognition). In FrameProcessor process method  app crash rarely and get null pointer exception from frame.getSize().getWidth();  Not always. What should i do? I added null control. But why that happens?

- CameraView version: 'com.otaliastudios:cameraview:1.6.1'
- Device / Android version: Nokia 8, 8.1 Oreo

### Code 
```

@Override
    public void process(@NonNull Frame frame) {
            width = frame.getSize().getWidth();
            height = frame.getSize().getHeight();
    ...}
```

closed 2019-02-24 by stale[bot]

renaudmathieu, 2019-03-11:
Having a similar issue with `2.0.0-beta04` → 
```java.lang.IllegalStateException: frame.size must not be null```

natario1, 2019-03-11:
@renaudmathieu can you post your code? I don't see how this could happen.

Only if you use the Frame after `onFrame` has returned.

renaudmathieu, 2019-03-11:
```
private val frameProcessor = FrameProcessor { frame ->
    if (frameMetadata == null) {
        frameMetadata = FirebaseVisionImageMetadata.Builder()
            .setWidth(frame.size.width)
            .setHeight(frame.size.height)
            .setFormat(frame.format) // CameraView always returns NV21 for now
            .setRotation(getSurfaceRotation(frame.rotation))
            .build()
    }
    frameMetadata?.let { metadata ->
        val data = frame.data
        val image: FirebaseVisionImage = FirebaseVisionImage.fromByteArray(data, metadata)
        detectInImage(image)
            .addOnSuccessListener { onSuccess(it) }
            .addOnFailureListener { e -> onFailure(e) }
    }
}
```