natario1/CameraView #371: memory leak problem

SungJoKim, 2019-01-25:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: 2.0.0-beta2
- Reproducible in official demo app: no
- Device / Android version: *Galaxy S9, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Enable leak canary
2. CameraFragment add, remove

### Expected behavior
a memory nick occurred.

### Screenshots

### Logs
- HandlerThread. uncaughtExceptionHandler
- Camera1.mCameraCallbacks
- CameraView$Callbacks.this$0
- CameraView.mParent
- ConstraintLayout.mParent
- ConstraintLayout

SungJoKim, 2019-01-25:
I called cameraView.destroy() on fragment.onDestory()
maybe, memory leak is  mCameraCallbacks is not null when called cameraView.destory()

natario1, 2019-01-28:
Please read instructions [here](https://natario1.github.io/CameraView/about/getting-started.html) : use `setLifecycleOwner(viewLifecycleOwner)` in fragments.

If you still want to call the callbacks manually, `camera.destroy()` must be called `onDestroyView()`, not `onDestroy()`.

fornewid, 2019-01-28:
Oh, I miscatched that.
I'll recheck my codes.
Thanks for rapid response. :)

SungJoKim, 2019-01-29:
When I told you to do it, I found that the problem was solved.
thanx
but i saw leak from fragment instance variables in camera frame callback

closed 2019-01-29 by SungJoKim