natario1/CameraView #372: Flash does not work

android4canada, 2019-01-25:
Hello
I am using camera view v2

If I set the flash to ON then it doesnot start the flash.

I went in the code and noticed that getSupportedControls is returning false although flash is supported and I was able to enable it using differ camera api. 

Any reason why? Could it be camera1 vs camera2?

I am using galaxy A8

android4canada, 2019-01-25:
If I don't use the cameraview and  I do this
```
CameraManager camManager = (CameraManager) getContext().getSystemService(Context.CAMERA_SERVICE);
                String cameraId = null; // Usually back camera is at 0 position.
                try {
                    cameraId = camManager.getCameraIdList()[0];
                    if (tbFlash.isChecked()) {
                        camManager.setTorchMode(cameraId, true);   //Turn ON
                    } else {
                        camManager.setTorchMode(cameraId, false);   //Turn off
                    }
                } catch (CameraAccessException e) {
                    e.printStackTrace();
                }
```

Then flash work. The work around I found is to do camerview.close() prior to calling the above code and then when I open right after, then flash is off. Is there anyways we can fix this bug to perhaps use the above the code or something? The problem that cameraview does not give me access to Camera object

closed 2019-02-24 by stale[bot]

androidcodehunter, 2020-10-14:
I am sure it's not working in MIA1 device. What will be the workaroung for this issue.