natario1/CameraView #375: Incorrect preview orientation when device  is rotated

mevinDhun, 2019-01-28:
I am trying to use this library in an app which needs to support both landscape and portrait orientations. The preview only respects the orientation it was launched in and gets rotated when the orientation is changed(see attached screenshots). But the image taken has the correct orientation. The workaround i tried is to i start and stop the camera whenever the device is rotated, then the preview will have the right display but this causes a crash if the device is rotated multiple times very quickly. I have this in my activity section:
```xml
 <activity
 android:name=".CameraActivity"
android:configChanges="orientation|screenLayout|keyboardHidden">
```

- CameraView version: *2.0.0-beta02*
- Reproducible in official demo app: *yes*
- Device / Android version: *Google Pixel 2, API 27*

### Steps To Reproduce
1. Remove android:screenOrientation="portrait" in AndroidManifest for demo app
2. Run on device and rotate to landscape

### Expected behavior
the preview should match the device orientation

### Screenshots
Portrait
[image]
Landscape
[image]

mevinDhun, 2019-02-14:
@natario1 have you been able to reproduce it? On my side, i added the following code to ```startPreview``` method on Camera1 class:
```java
       Camera.CameraInfo info = new Camera.CameraInfo();
        Camera.getCameraInfo(mCameraId, info);
        int displayOrientation = (info.orientation - this.mDeviceOrientation + 360) % 360;
        if (displayOrientation == 180)
            displayOrientation = 0;
        else if (displayOrientation == 0)
            displayOrientation = 180;
        mCamera.setDisplayOrientation(displayOrientation);
```
The full size preview now has the correct orientation when the device is rotated but it gets distorted/stretched. Any ideas why the preview size is not correctly being calculated?

natario1, 2019-02-23:
Thanks, after looking into it I don't think this is a bug. You just have a wrong manifest.

```xml
<activity
    android:name=".CameraActivity"
    android:configChanges="orientation|screenLayout|keyboardHidden">
```

If you want to work with both orientations, you must remove the `android:configChanges` line. Take a look at what that means in docs.

So either you use `android:configChanges` and `android:screenOrientation`, or you remove both of them.

mevinDhun, 2019-02-25:
@natario1 I do not want my activity to be restarted when there is an orientation change, so i need the ```configChanges```. But the app also needs to support both landscape and portrait, so i added ```fullSensor``` but it still has the incorrect preview.
```
android:configChanges="orientation|screenLayout|keyboardHidden"
android:screenOrientation="fullSensor
```

natario1, 2019-02-25:
OK I understand now what you meant, but if you list `orientation` in configChanges (and don't lock the orientation using `screenOrientation`) this means that you have to handle orientation changes yourself.

I think you can just close and reopen the camera during `onConfigurationChange`. Because this change leaves CameraView in a bad state which I don't think can be fixed without restarting

mevinDhun, 2019-02-25:
I had already tried that it worked to some extent but if you do a lot of rotations, the camera eventually crash (from the native android camera). Do you have any suggestion on how i can improve [the PR i submitted](https://github.com/natario1/CameraView/pull/386)?

natario1, 2019-02-25:
With what exception?

I can't accept an PR that modifies existing code about orientations, because existing code works and it was a long journey to get it right on most devices.

Since you use `configChanges=orientation` without locking orientation, what you have to do is find a way to restart the camera during `onConfigurationChange`.

The `CameraView` could have a new method called `onConfigurationChange()` which could help people like you to do this. But the implementation of this method, as I see this, should be `close(); open();` which is two lines so it's not really worth it.

natario1, 2019-02-25:
You can try this 

- on activity `onConfigurationChange` call `CameraView.onConfigurationChange`
- in `CameraView.onConfigurationChange`, try this

```java
mOrientationHelper.enable(getContext());
mCameraController.setDisplayOffset(mOrientationHelper.getDisplayOffset());
```

mevinDhun, 2019-02-27:
@natario1 i tried using the ```onConfigurationChange``` to update the display offset but it now crashes in the ```startPreview``` method on this line:
```
 mCamera.setParameters(params);
```
```
02-27 12:04:23.512 25523-25570/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewController
    Process: com.otaliastudios.cameraview.demo, PID: 25523
    java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:1898)
        at com.otaliastudios.cameraview.Camera1.startPreview(Camera1.java:203)
        at com.otaliastudios.cameraview.Camera1.access$700(Camera1.java:24)
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:78)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```
It seems the the preview size is not being calculated properly which makes the native camera reject the params

natario1, 2019-03-15:
So I guess we do not support this use case. I suggest using `close()` followed by `open()` in `onConfigurationChange()`.

closed 2019-03-29 by natario1