natario1/CameraView #378: Fatal Exception: java.lang.RuntimeException: setParameters failed

ankitbatra11, 2019-02-03:
### Describe the bug
Setting camera parameters failed.

```
Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:1964)
       at com.otaliastudios.cameraview.c$7.run(Camera1.java:823)
       at com.otaliastudios.cameraview.c$8.run(Camera1.java:63)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:145)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

- CameraView version: 1.6.0
- Reproducible in official demo app: no
- Device / Android version: Samsung Galaxy J3(2016)/ Android Version 5.1.1

The crash has been reported through Crashlytics. I can't reproduce locally. Logs aren't available too for the same reason.

Setting camera parameters should not fail with an exception.

natario1, 2019-02-03:
Thanks! Unfortunately I'm not planning to work on any improvement/fix to version 1.X.X. This is the kind of thing that might be fixed in v2. 

(I'm open to accept PRs where the issue has been identified - not random try catches - and release a 1.6.1 at some point)

If this issue was about v2, I also would not be able to do much about it without more info than this. Consider logging more info (for future bugs, or this same bug in v2) with CameraLogger.

nickleichliter93, 2019-02-03:
What is this

ankitbatra11, 2019-02-04:
I am waiting for 2.0.0-beta to be promoted before using that in my app. Go ahead and close this if you want.

natario1, 2019-02-09:
Will close. I took a look at the code lines and found no evident issue there, must be some device specific thing which I don't know how to address at the moment.

closed 2019-02-09 by natario1

fornewid, 2019-04-01:
My app is also detected this issue. 😢 

- CameraView version: 2.0.0-beta03
- Reproducible in official demo app: no
- Device / Android version: KYOCERA KYT31/ Android Version 5.1

natario1, 2019-04-01:
Hey @fornewid , can you open a new issue and fill the information that you have? Stack trace, steps to reproduce (when does this happen?), logs.