natario1/CameraView #379: Force recorded video rotation

2math, 2019-02-05:
### How do I?
Hi, is it possible to force the CameraView to record video in specific orientation, like Landscape for example?
I have my activity locked to landscape, but if the user continue to keep the device in portrait position, no matter that the content view is in landscape mode the recorded video is still in portrait. 
It seems like the CameraView uses the device's sensor.
Thanks in advance!

### Version used
com.otaliastudios:cameraview:1.6.1

natario1, 2019-02-05:
No, it's not possible, sorry. We use the device's sensor.

aartikov, 2019-02-07:
@natario1 
I need this feature for photos too. I copied library source code and changed OrientationHelper as a temporary solution. Is it possible to have some extension point here?

natario1, 2019-02-15:
Hey @aartikov , a PR would be appreciated!

There should be an API + xml attribute to enable and disable this behavior. Something like `setUseDeviceOrientation` and `cameraUseDeviceOrientation`?

dummy-amit, 2019-05-20:
> OrientationHelper

I want my recorded video to be always in portraint mode (zero rotation), doesn't matter how user rotates video. Could you please guide me where did you make change for that. Does making change only in orientation helper is enough?

PalakSDarji, 2019-07-02:
I too want to disable the automatic rotation applied on video output. @aartikov  Can you share with us OrientationHelper changes? Thanks!

huy-t, 2019-07-09:
Currently I use this lib for user capture their ID card. 
They often put it in table and capture.
Some time this lib is mixing between Portrait and Landscape mode.
So this function should be provide soon.

ManuMobsuccess13, 2019-07-11:
Same for me here , I'm stuck on a problem where my recording appears in landscape but everything else on my app is in portrait mode

aartikov, 2019-07-12:
@PalakSDarji
Sure. I have pushed my changes in this commit https://github.com/aartikov/CameraView/commit/a71c0f228c8136f929e044bc451cbc7b6334d401
I have added method `setAllowedOrientation` to `CameraView` with possible values: `LANDSCAPE`, `PORTRAIT`, `ANY`. You can check out how it works in the demo app.

@natario1
I don't make a PR with my solution because I am not sure that it will fit needs of everybody. What do you think about it? As I can see you already have a PR that addresses the issue. Sorry that I am too later with my comment.

ManuEpi, 2019-07-12:
Hello,

@natario1 
Thanks , when do you think it will be implemented on your projet and on which version please ?
I'm still using the 1.6.1 version actually :/

closed 2019-07-12 by natario1

natario1, 2019-07-12:
Hey, you can use setUseDeviceOrientation now (to be released in next version).

@ManuEpi time to migrate! :-) will go out of beta soon.

ManuEpi, 2019-07-12:
Hey ,  I will migrate for the next version which include "setUseDeviceOrientation" :)
Thank you

AdrianLeeElder, 2020-08-11:
@natario1 will this be available in a future release? I currently need this same feature.

PR #497 (2019-07-12): Use device orientation flag
- Fixes #379 
- Tests: yes
- Docs updated: yes