natario1/CameraView #382: How to remove RECORD_AUDIO permission from Manifest

therealshabi, 2019-02-09:
I don't want to keep `<uses-permission android:name="android.permission.RECORD_AUDIO"/>` permission in my Manifest plainly due to the fact that I am not using this in my app. But the project does not build if I don't give this permission. Do you have any solution for that?
### Version used
I am using `com.otaliastudios:cameraview:2.0.0-beta02` version of cameraview

natario1, 2019-02-09:
What do you mean that the project does not build?

therealshabi, 2019-02-09:
I meant App is crashing without any reason

natario1, 2019-02-09:
Try disabling audio with cameraAudio="off"

therealshabi, 2019-02-09:
No, I actually don't want to keep this permission in the Manifest itself, because I won't be using this.
Moreover, when I'm uploading my app to play store I have to agree to a privacy policy due to this `RECORD_AUDIO` permission which in my opinion is unnecessary

natario1, 2019-02-09:
I understand! Try doing what I said

therealshabi, 2019-02-09:
Where do I have to put this `cameraAudio = "off"` btw?

natario1, 2019-02-09:
In the XML definition of your CameraView, or call cameraView.setAudio(Audio.OFF) as soon as you can

therealshabi, 2019-02-09:
It's still crashing unfortunately

natario1, 2019-02-15:
With what stacktrace?

closed 2019-03-14 by stale[bot]

dickyabd, 2022-03-30:
Hi @natario1 we have same problem where the RECORD_AUDIO permission still in the app's manifest. Note that we've already following this step : https://github.com/natario1/CameraView/issues/382#issuecomment-462054261

if you can help us to solve this, really appreciated

dickyabd, 2022-03-31:
nevermind, we're able to remove the record_audio permission on our app.