natario1/CameraView #384: mPreview is null at Camera1.unbindFromSurface(150)

huseyinWA, 2019-02-11:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: *2.0.0-beta2*
- Reproducible in official demo app: *yes/no*
- Device / Android version: *Many Device and OS*

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Class com.otaliastudios.cameraview.h.b()' on a null object reference
       at com.otaliastudios.cameraview.Camera1.unbindFromSurface(Camera1.java:150)
       at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:278)
       at com.otaliastudios.cameraview.CameraController$4.run(CameraController.java:202)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2019-02-11:
Thanks, this should not happen

huseyinWA, 2019-02-11:
I couldn't reproduce but it crashes in production. It has occured after update to 2.0.0-beta2, before I haven't seen.

natario1, 2019-02-23:
I have added a check so that this does not crash, but this is an usage error, it seems that you are calling open/close when the CameraView object is not attached to any window (an activity or an attached fragment). So, calling open() too early.

commit d462b830 (2019-02-23): v2.0.0-beta03 (#392)

* Fix #377 and update dependencies

* Fix #384

* Fix unbindFromSurface bug

* Bump version to v2.0.0-beta03

* Update build.gradle
closed 2019-02-23 by natario1