natario1/CameraView #389: Camera Front facing camera preview dark on video

pankaj89, 2019-02-19:
### How do I?
When taking video from front facing camera, camera preview become darker. when i toggle facing preview become normal, I checked with device camera, device camera works perfect.

I am checking in 
Device - Pixel 1
Android Version - 9

Is there any special parameter i need to add? or it some kind of issue?  Please help.

### Version used
com.otaliastudios:cameraview:2.0.0-beta02

closed 2019-03-18 by stale[bot]

BlindJerobine, 2019-06-26:
Same issue with my Pixel 2XL & some other test devices.

Preview stays light till video starts. While the video is recorded, it's extremely dark. The Output is dark, too!

Version used: 2.0.0-beta05

TemMax, 2019-12-02:
Faced with same issue on my pixel 2

version used: 2.4.0

TemMax, 2019-12-02:
@natario1 may be you can help with this issue or reopen?

yaroslavkulinich, 2020-03-26:
The same with Pixel 1.

mariuspena, 2020-05-06:
please fix this

TemMax, 2020-06-16:
If you don't need really hq video output just use CameraView#takeVideoSnapshot. With recording from surface this trouble isn't reproduced.