natario1/CameraView #390: The demo app freezes after recording a video on Nexus 5

hainkyu, 2019-02-22:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: v2.0.0-beta02
- Reproducible in official demo app: yes
- Device / Android version: Nexus 5 / 6.0.1

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. From the demo app, change to Video mode
2. Start recording a video 
3. After 6 sec (finished video recording), the app freezes (the same issue happens with v1.6, not only with the latest v2 beta release)

### Expected behavior
The app shouldn't freeze

### Logs
`02-22 12:43:47.785 19948-20023/com.otaliastudios.cameraview.demo W/MediaRecorder: mediarecorder went away with unhandled events
02-22 12:43:47.795 19948-20023/com.otaliastudios.cameraview.demo E/Camera: Error 1
02-22 12:43:47.795 19948-20023/com.otaliastudios.cameraview.demo E/Camera1: Error inside the onError callback. 1
02-22 12:43:47.798 19948-20023/com.otaliastudios.cameraview.demo E/CameraController: uncaughtException: Interrupting thread with state: STATE_STARTED due to CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 1
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 1
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:335)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)
     Caused by: java.lang.RuntimeException: Error inside the onError callback. 1
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:328)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146) 
        at android.os.Handler.dispatchMessage(Handler.java:102) 
        at android.os.Looper.loop(Looper.java:148) 
        at android.os.HandlerThread.run(HandlerThread.java:61) `

### APK
Link to a Github repo where the bug is reproducible.

natario1, 2019-02-22:
You should handle errors in `onCameraError` inside your callback. Something happened and the error will contain some information

hainkyu, 2019-02-22:
@natario1 Thanks for your comment.
I've already been handling the exception (trying to open the camera again).
The issue is more about the app becoming frozen each time after recording a video.
Btw, it may be a device specific issue as I don't see the problem with some other devices.

natario1, 2019-02-23:
I don't know! It is an internal camera error by the Camera1 error callback, it's out of our control. If I had the device at hand I could do more testing, but I don't, so, this is not actionable for me. If you can spot an error, feel free to comment here

closed 2019-03-15 by natario1