natario1/CameraView #391: Issue with device orientation and image rotation

edgars-supe, 2019-02-22:
### Describe the bug
There are multiple scenarios in which the library calculates the device orientation erroneously, resulting in incorrectly rotated pictures (and in some cases probably causes the preview to be rotated incorrectly as well).
1) After shaking the device in such a way that the display/activity orientation does not change, it's possible to trick the library into thinking it has rotated. 
2) After shaking the device in such a way that it does trigger an orientation change, but quickly putting it back as it was (without the OS triggering a configuration change), the library does not seem to "reset" its orientation (or something). I've observed, that after changing the orientation of the device properly, `onOrientationChanged` gives me, for example, `90` and then `0` immediately afterwards. When shaking it, it just goes to `90`. If I tilt my phone up later, then it gives me `0` as well. 

- CameraView version: *2.0.0-beta02*
- Reproducible in official demo app: *yes*
- Device / Android version: *Sony Xperia XZ1 Compact, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Face device downwards (parallel to the ground, in portrait)
2. Quickly shake the device as if you're turning it to landscape mode, but go back to as it was.
3. Take picture.

OR

1. Face device downwards (in portrait)
2. Quickly shake the device so that it triggers an orientation change, but keep it facing downwards (activity is now in landscape)
3. Take picture.

### Expected behavior
Resulting picture is oriented correctly, as seen in the preview.

### Screenshots
[Phone held parallel to the table](https://drive.google.com/open?id=1fH7acZCa__xrNM4hCKk1wqTeNJrKR7V3).
[After shaking the phone](https://drive.google.com/open?id=1DORHvewccHGlW-8grXZ4DQeqaGLOh01q).

[Here's a video](https://drive.google.com/open?id=1p9nJEQU4s5pOEVcZ7tqMvhEmJ2EXeGfd). The number in the middle of the screen is the rotation as reported by `CameraListener#onOrientationChanged`. 

### Logs
After shaking the device, the orientation is reported as changed:
`2019-02-22 18:01:28.669 23807-23807/[redacted] I/CameraCallbacks: onDeviceOrientationChanged 270`
And [this is after actually putting the device in landscape](https://drive.google.com/file/d/1rAChHAg3ejkmsxhUGzBMpB-Y0FSB-S33/view?usp=sharing).

natario1, 2019-02-23:
Thanks for your report, but pictures taken with device parallel to the ground have no meaningful orientation, there's no good and bad orientation.

edgars-supe, 2019-02-23:
I do realize that. The reason I wanted to switch over to your library for our project was that it supports taking a snapshot of the preview instead of taking a picture, because, as you said, parallel to the ground, there's no landscape or portrait. And that's very bad for our use-case, if the user has their phone oriented in landscape, but pointing downwards - the EXIF rotation data, which we use to rotate the pictures to a "normal" state, is misleading/incorrect, resulting in improperly cropped images. 

Anyway, I must admit, that I forgot to mention this in the bug report, but, yeah, as I mentioned, I use `takePictureSnapshot()` to capture the images (and also pressed the "Snap" button in the demo app, not even realizing that it's specifically for snapshots and the button to the left is for regular pictures). And it works great even in landscape and parallel to the ground, granted you're in the landscape orientation and haven't shook your phone around. The problem appeared to be solved, but then we discovered this bug. 

Is there a reason to use orientation data in `snapshot` mode? If it takes the image directly from the preview, does it not just read data off the `SurfaceView` or `TextureView`, or whatever is there? Or does it not work like that? Because, if it does work like I described, I don't see a reason to take into account orientation data.

natario1, 2019-02-23:
@edgars-supe no matter the capture type (picture, picture snapshot, video, video snapshot) the library tries to provide output that is consistent to what the user was seeing.

Since the app does not know the user position, it works with some assumptions:
- phone is somewhat vertical, & we need the device orientation
- user is vertically standing as well
This way we can reconstruct their position in space and this works 99% of the times.

There has been some discussion in #379 about adding a flag to tell CameraView to ignore the device orientation. You can check out with people there and maybe work on it?

edgars-supe, 2019-02-25:
I understand. I'll see if I can override this behaviour for our project. Thanks!

closed 2019-02-25 by natario1