natario1/CameraView #397: java.io.IOException: setPreviewTexture failed

artemkopan, 2019-02-28:
### To Reproduce
Steps to reproduce the behavior:
1. Create two fragments with cameraview
1. Open the first fragment
2. Take a photo and then open the second fragment (also with CameraView)

PS. Tried with different cameraPreview modes

### Logs
```
02-22 22:03:20.036 1479-1479/com.accelior.next.user.dev E/Camera1: unbindFromSurface Could not release surface java.io.IOException: setPreviewTexture failed
    java.io.IOException: setPreviewTexture failed
        at android.hardware.Camera.setPreviewTexture(Native Method)
        at com.otaliastudios.cameraview.Camera1.unbindFromSurface(Camera1.java:162)
        at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:287)
        at com.otaliastudios.cameraview.CameraController.stopImmediately(CameraController.java:217)
        at com.otaliastudios.cameraview.CameraController.destroy(CameraController.java:155)
        at com.otaliastudios.cameraview.CameraView.destroy(CameraView.java:685)
        at com.accelior.next.presentation.features.kyc.photo.KycPhotoFragment.onDestroyView(KycPhotoFragment.kt:49)
        at com.accelior.next.presentation.features.kyc.photo.profile.KycProfilePhotoFragment.onDestroyView(KycProfilePhotoFragment.kt)
        at androidx.fragment.app.Fragment.performDestroyView(Fragment.java:2770)
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:960)
        at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1233)
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1306)
        at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1874)
        at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1820)
        at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1721)
        at androidx.fragment.app.FragmentManagerImpl$1.run(FragmentManagerImpl.java:147)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
02-22 22:03:20.041 1479-1479/com.accelior.next.user.dev E/libEGL: call to OpenGL ES API with no current context (logged once per thread)
02-22 22:03:20.056 1479-1624/com.accelior.next.user.dev E/Camera: Error 2
02-22 22:03:20.056 1479-1624/com.accelior.next.user.dev E/Camera1: Error inside the onError callback. 2
```

natario1, 2019-02-28:
You have to make sure that destroy() in the first fragment is called before open() in the second fragment

closed 2019-03-15 by natario1

olashiku, 2019-06-14:
this method does not work after using the specified call back I got this error. what I was trying to achieve is to go back to my previous fragment with the camera view still active. 

06-14 15:44:45.320 7651-7671/com.qucoon.testingcamerapreview E/Camera1: unbindFromSurface Could not release surface java.io.IOException: setPreviewTexture failed
    java.io.IOException: setPreviewTexture failed
        at android.hardware.Camera.setPreviewTexture(Native Method)
        at com.otaliastudios.cameraview.Camera1.unbindFromSurface(Camera1.java:163)
        at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:290)
        at com.otaliastudios.cameraview.CameraController.stopImmediately(CameraController.java:224)
        at com.otaliastudios.cameraview.CameraController.destroy(CameraController.java:162)
        at com.otaliastudios.cameraview.CameraController.uncaughtException(CameraController.java:123)
06-14 15:44:45.320 7651-7651/com.qucoon.testingcamerapreview E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.qucoon.testingcamerapreview, PID: 7651
    java.lang.RuntimeException: getParameters failed (empty parameters)
        at android.hardware.Camera.native_getParameters(Native Method)
        at android.hardware.Camera.getParameters(Camera.java:1996)
        at com.otaliastudios.cameraview.Camera1.startPreview(Camera1.java:185)
        at com.otaliastudios.cameraview.Camera1.access$700(Camera1.java:24)
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:80)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:61)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)