natario1/CameraView #398: Not working method takePicture on SENSEIT E510

mrtsmglv, 2019-02-28:
### Describe the bug
When I try take picture from main camera nothing happens, but take picture from front camera it's work.

- CameraView version: *2.0.0-beta03*
- Reproducible in official demo app: *yes*
- Device / Android version: *SENSEIT E510/6.0*

### To Reproduce
Steps to reproduce the behavior, in the demo app:
1. Go to demo app
2. Click toggleCamera
3. Click on capture picture (R.id.capturePicture)
4. Opening preview
5. Back pressed
6. Click toggleCamera
7. Click on capture picture (R.id.capturePicture)
8. Notihng
 

### Expected behavior
Opening window of picture preview

### Screenshots
[image]

### Logs
[Here](https://gist.github.com/mrtsmglv/0fb89aa6597f836729acefea21db565e)

natario1, 2019-03-15:
Thanks for the good report, but I don't see anything wrong in these logs. So I don't think I can help without having this specific device at hand.

See if you can spot any issue with the debugger. Feel free to reopen in that case.

closed 2019-03-15 by natario1

HBiSoft, 2019-04-01:
I have actually had a few users report this to me, I can't log the issue as it only happens on some devices.

@mrtsmglv can you please provide only the part of your log from where you press capture?

mrtsmglv, 2019-04-01:
@HBiSoft Unfortunately, this device was taken away from me and I cannot get it :(

HBiSoft, 2019-04-01:
@mrtsmglv No problem, I placed a log so I can catch it in my next release, thank you.