natario1/CameraView #399: Migration issue

HassanUsman, 2019-03-01:
### How do I?
I've updated the library from version 1.5.1 to 2.0.0-beta03 and facing this issue. 

[image]

I'm using in android.app.Fragment and caling this Fragment from AppCompatActivity I'm not using androidx.lifecycle library if I import this library than issue is resolved but when I take picture app crashed. I tried many solutions but not able to resolve this.

### Version used
'com.otaliastudios:cameraview:2.0.0-beta03'

natario1, 2019-03-01:
I think you need to add androidx to your project, see #351

HassanUsman, 2019-03-02:
ooh, no man that's a very hectic job. Is there any other way except this one.

natario1, 2019-03-02:
I have no idea as I have never tried. There is some discussion there

HassanUsman, 2019-03-03:
Ok, Thanks.

HassanUsman, 2019-03-04:
@natario1 one more question. The picture quality isn't good how I enhance it.

natario1, 2019-03-04:
If you are using takePictureSnapshot, use takePicture instead, if needed control the size using setPictureSize and take a look at the documentation

HassanUsman, 2019-03-05:
@natario1 I already using `takePicture() `method and I applied the selector also but it didn't show the expected result here is my code
```
camera = findViewById(R.id.camera);
        camera.setLifecycleOwner(this);
        SizeSelector width = SizeSelectors.minWidth(1280);
        SizeSelector height = SizeSelectors.minHeight(720);
        SizeSelector dimensions = SizeSelectors.and(width, height); // Matches sizes bigger than 1000x2000.
        SizeSelector ratio = SizeSelectors.aspectRatio(AspectRatio.of(4, 3), 0); // Matches 1:1 sizes.

        SizeSelector result = SizeSelectors.or(
                SizeSelectors.and(ratio, dimensions), // Try to match both constraints
                ratio, // If none is found, at least try to match the aspect ratio
                SizeSelectors.biggest() // If none is found, take the biggest
        );
        camera.setPictureSize(result);
```
And its make a `378x504` resolution image file. Please tell me hoe I acheive my desired result.

HassanUsman, 2019-03-06:
@natario1 waiting for your reply.

closed 2019-03-29 by natario1