natario1/CameraView #400: How do I reduce the resolution of Frame Processor?

GianfrancoMS, 2019-03-01:
### How do I reduce the resolution of Frame Processor?
I'm using CameraView with Firebase ML to read barcodes in real-time. Firebase recommends 480x360 to do real-time processing. How can I reduce the resolution of Frame Processor to 480x360?

Currently, the code for my frame processor is this:
```
        val frameProcessor = FrameProcessor { frame ->
            if (frame != null && frame.data != null && frame.size != null) {
                val now = System.currentTimeMillis()
                if (now - lastFrameTimeElapsed < previewRate) {
                    return@FrameProcessor
                }

                lastFrameTimeElapsed = now

                val frameMetadata = FrameMetadata(
                    width = frame.size.width,
                    height = frame.size.height,
                    rotation = frame.rotation / 90,
                    cameraFacing = 0
                )

                val byteBuffer: ByteBuffer = ByteBuffer.wrap(frame.data)
                barcodeScanningProcessor.process(byteBuffer, frameMetadata, overlay)
            }
        }
```

And my activity looks like this:

```
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:keepScreenOn="true"
        app:cameraAudio="off"
        app:cameraPlaySounds="false"
        //I don't why but surface works better for real-time processing.
        app:cameraPreview="surface">

        <com.enfasys.kippu.features.barcode.GraphicOverlay
            android:id="@+id/graphicOverlay"
            android:layout_width="match_parent"
            android:layout_height="match_parent" />

    </com.otaliastudios.cameraview.CameraView>
```

```
    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)

        //Other code

        camera = findViewById(R.id.camera)
        camera.setLifecycleOwner(this)

        //Other code
    }
```
### Version used
2.0.0-beta03

GianfrancoMS, 2019-03-01:
I tried this, but I didn't work. The width and height I'm getting are 1920 and 1080, respectively.

```
        camera.setPreviewSize(
            SizeSelectors.and(
                SizeSelectors.minHeight(360), SizeSelectors.maxHeight(360),
                SizeSelectors.minWidth(480), SizeSelectors.maxWidth(480)
            )
        )
```

GuanacoDevs, 2019-03-02:
Try this with CameraView2:
```
   cameraView.setPreviewSize(getSize());

   private SizeSelector getSize() {
        SizeSelector width = SizeSelectors.minWidth(480);
        SizeSelector sizeSelector = SizeSelectors.smallest();
        return SizeSelectors.and(width, sizeSelector);
    }
```
Note however that this ussually returns `480x640`
And I have a View(ViewFinder like) 180dpx180dp where the user must align the QRCode, in my case, those dimensions guarantee me at least 240px*240px resolution for the Code to be read.

With cameraView 1 you could do:
```
        app:cameraPictureSizeMinWidth="480"
        app:cameraPictureSizeSmallest="true"
```

closed 2019-03-02 by GianfrancoMS