natario1/CameraView #405: Face Detection with MLKit

bigahega, 2019-03-09:
### How do I?
I am trying to detect faces using Firebase MLKit and CameraView. However; as far as I have been able to debug there is something wrong with image formats.

I have the following firebase vision image metadata:

```
private val metaData = FirebaseVisionImageMetadata.Builder()
            .setWidth(480)
            .setHeight(360)
            .setFormat(FirebaseVisionImageMetadata.IMAGE_FORMAT_NV21)
            .build()
```

and within the frameProcessor callback I am creating the `FirebaseVisionImage` as the following:
`FirebaseVisionImage.fromByteArray(frame.data, visionImageMetadata)`

When I check the image with the `bitmapForDebugging` this is the following output:
[image]

What am I missing here?

### Version used
com.otaliastudios:cameraview:2.0.0-beta04

Priyanka-Mohanty, 2019-03-12:
did you find any solution?

natario1, 2019-03-13:
You must use frame.size.width, frame.size.height and frame.size.format to build this object, not your own width/height

Priyanka-Mohanty, 2019-03-13:
can you please share code???

natario1, 2019-03-13:
I don't use Firebase ML so I don't really know. But you have to read the `Frame` properties.

Priyanka-Mohanty, 2019-03-13:
private fun setupCamera() {
        val options =
            FirebaseVisionFaceDetectorOptions.Builder().setPerformanceMode(FirebaseVisionFaceDetectorOptions.ACCURATE)
                .setLandmarkMode(FirebaseVisionFaceDetectorOptions.ALL_LANDMARKS).build()
        val detector = FirebaseVision.getInstance()
            .getVisionFaceDetector(options)
        cameraView.addFrameProcessor {
           var frame= Frame(
                data = it.data,
                rotation = it.rotation,
                size = Size(it.size.width, it.size.height),
                format = it.format,
                isCameraFacingBack = cameraView.facing == Facing.FRONT)

            val faces  = faceDetector.process(frame)

            val frameMetadata = FirebaseVisionImageMetadata.Builder()
                .setWidth(it.size.width)   // 480x360 is typically sufficient for
                .setHeight(it.size.height)  // image recognition
                .setFormat(it.format)
                .build()

            val image = FirebaseVisionImage.fromByteArray(it.data, frameMetadata)

            detector.detectInImage(image).addOnSuccessListener {results ->

                val resultSize = results.size
                if (resultSize > 0) {

                    for (i in results.indices) {
                        val face = results[i]
                        var rightEyeOpenProb: Float = 1f
                        var leftEyeOpenProb: Float = 1f

                        if (face.rightEyeOpenProbability != FirebaseVisionFace.UNCOMPUTED_PROBABILITY) {
                            rightEyeOpenProb = face.rightEyeOpenProbability
                            System.out.println("right EyeOpenProb =$rightEyeOpenProb")
                        }
                        if (face.leftEyeOpenProbability != FirebaseVisionFace.UNCOMPUTED_PROBABILITY) {
                            leftEyeOpenProb = face.leftEyeOpenProbability
                            System.out.println("left EyeOpenProb =$leftEyeOpenProb")
                        }

                        if (rightEyeOpenProb <= 0.1 && leftEyeOpenProb <= 0.1) {
                            //eyes are close
                            Log.d(TAG, "Eyes are close!!!")

                        } else {
                            Log.d(TAG, "Please close your eyes!!!")
                            /*    DynamicToast.make(activity,"Please close your eyes!!!",
                                   mContext.getDrawable(R.drawable.ic_info) , mContext.getColor(android.R.color.holo_blue_dark), mContext.getColor(android.R.color.white)).show()
                            */
                        }
                    }

                    val face = results[0]
                    val faceRect = face.boundingBox
                    var bitmapImage = image.bitmapForDebugging
                    Log.e("TAG" ,"faceRect $faceRect")

                }
            }.addOnFailureListener{error->
                Toast.makeText(applicationContext, error.message, Toast.LENGTH_SHORT).show()
            }

        }

    }

sandeshsk, 2019-03-18:
Hello Currently what you need to achieve?

Priyanka-Mohanty, 2019-03-18:
> Hello Currently what you need to achieve?

i want to implemented face detection with ML Kit....

sandeshsk, 2019-04-08:
Sorry for late reply. You can achieve it but currently it is slow I have tried. you dont need  to use image.bitmapfordebug.
cameraview only has picture taken callback in which you can store file or you can get the bitmap also.

You can achieve different concept using eulery angle to detect real face please have a look at this tutorial.
https://medium.com/onfido-tech/face-detection-and-tracking-on-android-using-ml-kit-part-1-fbee4200d174

closed 2019-05-05 by stale[bot]

korbi-oranssi, 2019-09-16:
Hello! I have faced a similar issue when using the method FirebaseVisionImage.fromByteArray(). Here is the code I'm using

`FirebaseVisionImageMetadata metadata = new FirebaseVisionImageMetadata.Builder()`
`                .setWidth(240)`
`                .setHeight(320)`
`                .setFormat(FirebaseVisionImageMetadata.IMAGE_FORMAT_NV21)`
`                .setRotation(FirebaseVisionImageMetadata.ROTATION_270)`
`                .build();`
`final FirebaseVisionImage firebaseImage = FirebaseVisionImage.fromByteArray(byteArray, metadata);`

- byteArray is the output I receive from the camera when a picture is taken.
- height and width are the picture resolution I am using.

If I try to decode the byteArray into a Bitmap using `BitmapFactory.decodeByteArray(byteArray, 0, byteArray.length)` I receive the expected image:

[image]

However, the `firebaseImage.getBitmap()` returns nonsense:

[image]

Recognition from a bitmap using the `FirebaseVisionImage.fromBitmap()` works perfectly. But I'm hoping to get it working with the byte array to not reduce the performance.

Tried passing different sized and rotations. Still no luck. Would be grateful for any hints.

ramvdixit, 2019-12-20:
FirebaseVisionImageMetadata metadata = new FirebaseVisionImageMetadata.Builder()
                        .setFormat(frame.getFormat()) // This is always 17 - FirebaseVisionImageMetadata.IMAGE_FORMAT_NV21
                        .setWidth(frame.getSize().getWidth())
                        .setHeight(frame.getSize().getHeight())
                        .setRotation(FirebaseVisionImageMetadata.ROTATION_270)
                        .build();
                FirebaseVisionImage firebaseVisionImage = FirebaseVisionImage.fromByteArray(frame.getData(), metadata);

dinesh1301, 2021-01-07:
@bigahega @Priyanka-Mohanty I guess the trick we were missing was scaling and accounting for imageFlip. While if the preview is full screen then scaling should cause much of the problem. The frame size might not actually match with the screen/preview size hence the coordinates returned by the mlkit might not fit exactly onto your screen. On top of it, if your camera is facing in reverse direction, the imageFlip happens, hence the ```x``` coordinate actually becomes ````screenWidth - x```` 
After hustling through various SO and github links I found the answer in sample code of MLVision library. I have modified it a bit to make it work for me.

``` kotlin
/**
 * This converts the image coords to screen coords
 *
 * @param imageCoords: Coordinates as returned by the mlkit library
 * @param imageHeight: Height of the frame
 * @param imageWidth: Width of the frame
 * @param screenWidth: Width of the screen
 * @param screenHeight: Height of the screen
 * @param isFlipped: Is the camera reversed 
 * @return
 */
fun convertToScreenCoords(
    imageCoords: PointF,
    imageHeight: Int,
    imageWidth: Int,
    screenWidth: Int,
    screenHeight: Int,
    isFlipped: Boolean
): PointF {
    val scaleX = screenHeight.toFloat()/imageWidth.toFloat()
    val scaleY =  screenWidth.toFloat()/ imageHeight.toFloat()
    return PointF( if (isFlipped) screenWidth -  imageCoords.x*scaleX else imageCoords.x*scaleX, imageCoords.y*scaleY)
}
```