natario1/CameraView #409: Does CameraView come with output image type and quality option ?

ElyDantas, 2019-03-14:
Does CameraView come with output image type and quality option ?

Common way:

```
Camera camera = Camera.open();
Parameters cameraParameters = camera.getParameters();
cameraParameters.setPictureFormat(ImageFormat.JPEG);
cameraParameters.setJpegQuality(50);
camera.setParameters(cameraParameters);
```

CameraView way:

`SizeSelectors sizes...
camera.setPictureSize(sizes);
`

natario1, 2019-03-14:
Format is always JPEG and quality is always the default quality (not sure which one, try reading Android docs).

But you can use takeSnapshot() to take fast, low quality jpegs.

ElyDantas, 2019-03-15:
Yes @natario1, i did take a look in takeSnapshot() but my preview's size is 1dp x 1dp hidden in main activity, that setup outputs really low quality photos.

User has a slider to choose between 0..100 for image capture quality, in that case i had to create a bitmap from `onPictureTaken{ result.getData() }` raw data and compress it `bitmap.compress(CompressFormat.JPEG, loggedUser.getJpegQuality(), stream);` to get desired result.

natario1, 2019-03-15:
Yes, that's what you have to do if you want to respect the quality slider.

(You can have a 1dp CameraView but still take bigger snapshots, just use `setPreviewStreamSize()` to decide the snapshot size)

closed 2019-03-29 by natario1