natario1/CameraView #410: How to start recording without initial user input

Flashspeed, 2019-03-15:
### 
How can start a video recording when a Fragment/Activity loads; without having to implement the `takeVideo()` and `stopVideo()` methods in a button click listener for example?

I'am finding it difficult to locate where in a Fragment life-cycle to start a video recording with the `takeVideo()` method. The Fragment onResume() method seemed the most obvious place to do so but I've not had success. 

I had a look at this [question](https://github.com/natario1/CameraView/issues/407) I thought was related
Any suggestions would be great, thanks :smile: 

### Version used
CameraView V2 (2.0.0-beta04)

natario1, 2019-03-15:
Take the video when the camera opens - that is, in the onCameraOpened callback.

Flashspeed, 2019-03-15:
> Take the video when the camera opens - that is, in the onCameraOpened callback.

Thanks, that fixed things.

closed 2019-03-15 by Flashspeed