natario1/CameraView #412: java.lang.RuntimeException:

chiyan171, 2019-03-18:
版本：implementation 'com.otaliastudios:cameraview:1.5.1'
机型：Mate 20 Pro (HWLYA) 、P20 Pro (HWCLT)
系统：Android 9

java.lang.RuntimeException: 
  at android.hardware.Camera.<init> (Camera.java:568)
  at android.hardware.Camera.open (Camera.java:398)
  at com.otaliastudios.cameraview.Camera1.onStart (Camera1.java:172)
  at com.otaliastudios.cameraview.CameraController$3.run (CameraController.java:180)
  at android.os.Handler.handleCallback (Handler.java:891)
  at android.os.Handler.dispatchMessage (Handler.java:102)
  at android.os.Looper.loop (Looper.java:207)
  at android.os.HandlerThread.run (HandlerThread.java:65)

HBiSoft, 2019-03-22:
This is the same issues as #189 I would first update the version that you are using to `implementation 'com.otaliastudios:cameraview:2.0.0-beta03'` you can also catch the exception and with #265

If you are using crashlytics, you can catch the exception and send it to crashlytics using CameraException.getReason(). By doing this you will be able to see what is causing the crash.

HBiSoft, 2019-03-22:
Are you experiencing this on your own device or do you get the crash on a users device (google play..)?

natario1, 2019-03-23:
I have no time to look into v1 issues or fix them, sorry! Please upgrade to v2 like @HBiSoft said

closed 2019-03-29 by natario1

HBiSoft, 2019-05-18:
@natario1 
Same as I described in #425 It's not related to V1 only. 

Since posting the above, I sent all the logs to crashlytics and posted it in #425 as well as how I implemented everything.