natario1/CameraView #414: It is found that the recording will be delayed for 2 seconds.

dalong982242260, 2019-03-18:
First of all, thank you for your library. It's very useful. Now there's a problem that when we call takeVideo () to record video, we find that it will delay recording for 2 seconds. I always lack the original 2S video. I don't know what's going on.

### Version used
compile 'com.otaliastudios:cameraview:2.0.0-beta04'

natario1, 2019-03-23:
I don't know either, but some delay is expected as there's a lot of things going on - posting your request to the camera thread, setting up the recorders and their own threads, launching the codec and finally starting the recording itself.

closed 2019-04-17 by dalong982242260

saadbzu, 2019-10-31:
I also face this problem. Can anyone have a solution?